ros/xacro #45: cmake: xacro_add_xacro_file() creates empty file on failure

xqms, 2015-01-18:
If one uses the xacro_add_xacro_file() macro to generate URDF models from xacro at build-time, and the xacro file contains a syntax error (or xacro fails for some other reasons), the system still produces an empty output file. On the next build iteration (assuming the xacro document has not changed), make sees this file (which has newer timestamp than the source) and will not attempt to re-run xacro.

This can lead to confusion, which is why we have switched to invoking xacro on-the-fly using roslaunch to get the URDF model.

A fix would be to move the file opening right to the end of xacro, when we can be reasonably sure that everything is going to be okay, or to remove left-over files in the `add_custom_command` call in CMake by some construct like this:

```
add_custom_command(OUTPUT ${output}
COMMAND ${_xacro_py} ${input} > ${output} || rm ${output}
DEPENDS ${input} ${_xacro_deps_result})
```

I haven't tested that, though.

codebot9000, 2015-01-19:
hi xqms,

thank you for the detailed bug report and proposed direction. I created a test case which replicates this failure mode, and then implemented a fix as you suggested.

-Morgan

closed 2015-01-19 by codebot9000

xqms, 2015-01-19:
Hi, thanks for the quick reaction!

I just noticed that `cmake/xacro-extras.cmake.em` uses the shell pipe syntax `> file` instead of `-o file`. Could you also change that so that packages using the macro can also take advantage of your fix?

codebot9000, 2015-01-24:
Hi,
I'm not sure why it was using shell redirection before. The flag will be more portable anyway (e.g., to Windows someday). I just committed your suggestion; give it a try and see if it works as expected for you. If all looks well, I'll push out a patch release.
-mq

commit 4c95467c (2015-01-19): fix #45. create output file after parsing is complete.