ros/xacro #49: Feature request: index support into bundled params

freyzor, 2015-02-15:
Trying to write modular xacto definitions I find myself really wanting to be able to index bundled xyz or rpi params passed along into macros.  This would greatly simplify making components.

To give an example here is a bioloid F2 bracket part used to compose composit links.
```xml
    <macro name="geometry_F2">
        <geometry>
            <mesh filename="package://${PACKAGE}/meshes/F2.stl" scale="${M_SCALE_3}"/>
        </geometry>
    </macro>
    <macro name="visual_F2" params="xyz rpy">
        <visual>
            <origin xyz="${xyz}" rpy="${rpy}" />
            <geometry_F2 />
            <color_grey60 />
        </visual>
    </macro>
    <macro name="collision_F2" params="xyz rpy">
        <collision_box xyz="${xyz}" rpy="${rpy}" size="0.025 0.0485 0.0375" />
    </macro>
    <macro name="part_F2" params="xyz rpy">
        <collision_F2 xyz="${xyz}" rpy="${rpy}" />
        <visual_F2 xyz="${xyz}" rpy="${rpy}" />
    </macro>
```
I then use it in to create a link like so
```xml
    <macro name="coxa_bone" params="parent name">
        <link name="${name}_link">
            <inertia_default />
            <part_ax12 xyz="0 0 0" rpy="0 0 ${MPI}" />
            <part_F3 xyz="0 0 -0.0415" rpy="0 ${MPI} 0 "/>
            <part_F2 xyz="0 0 -${L_COXA}" rpy="0 0 ${MPI2} "/>
        </link> 
        <joint_revolute name="${name}" parent="${parent}" xyz="0 0 0" rpy="0 0 0" 
            vel="100" llimit="${MIN_RAD_COXA}" ulimit="${MAX_RAD_COXA}" axis="0 1 0" />   
    </macro>
```

If I want to provide an offset to the xyz in the collision_F2 macro as it does not align to the stl mesh I'm unable to extract the x y z components again to add an offset.  You can get around this by passing in the values separately but that does brake the encapsulation and makes the whole thing pretty verbose again. 

what I'd like to be able to do would be something like the following.
```xml
    <macro name="collision_F2" params="xyz rpy">
        <collision_box xyz="${xyz[0]+0.03 xyz[1]-0.4 xyz[3]}" rpy="${rpy}" size="0.025 0.0485 0.0375" />
    </macro>
```
This way I can keep the implementation detail neatly contained.

rhaschke, 2015-02-18:
I have worked on a [python-based evaluation of expressions](https://github.com/ubi-agni/xacro), which allows to do what you want:
```xml
<robot name="myRobot"
       xmlns:xacro="http://ros.org/wiki/xacro">
	<macro name="index" params="xyz rpy">
		<property name="xyzList" value="${map(float, xyz.split(' '))}"/>
		<box xyz="${' '.join(map(str, [xyzList[0]+1, xyzList[1]-2, xyzList[2]]))}"
		     rpy="${rpy}"/>
	</macro>
	<index xyz="1 2 3" rpy="1 2 3"/>
</robot>
```
results in
```xml
<robot name="myRobot" xmlns:xacro="http://ros.org/wiki/xacro">
  <box rpy="1 2 3" xyz="2.0 0.0 3.0"/>
</robot>
```

As you see, you need to convert the string to a list first and then back to a string.

rhaschke, 2015-02-18:
There is even a slightly simpler method to realize it:
```xml
<robot name="myRobot"
       xmlns:xacro="http://ros.org/wiki/xacro">
	<macro name="index" params="xyz rpy">
		<property name="xyzList" value="${map(float, xyz.split(' '))}"/>
		<box xyz="${xyzList[0]+1} ${xyzList[1]-2} ${xyzList[2]}"
		     rpy="${rpy}"/>
	</macro>
	<index xyz="1 2 3" rpy="1 2 3"/>
</robot>
```

freyzor, 2015-02-19:
excellent, looks really good.
I this should open up interesting macro creation I guess like
```xml
<macro name="as_float_list" params="name source">
    <property name="${name}" value="${map(float, ${source}.split())}" />
</macro>
```

freyzor, 2015-02-19:
I tried this out on my a rather complicated setup and I ran into a curious issue where a xyz attribute with a bunch of calucated values suddenly got truncated into a single float.  This was not using any of the python capabilities specifically.  I'll have investigate to see if I can figure out where this is failing.

rhaschke, 2015-02-19:
If you like, send me a minimal code example that fails.

-- 
Robert Haschke

--------------------------------------------------------------------------
Dr. Robert Haschke
Faculty of Technology, Neuroinformatics Group, CITEC-3.309
Bielefeld University, 33615 Bielefeld, Germany
phone:  +49-521-106 12122
fax:    +49-521-106 6011
mobile: +49-157-817 365 13
WWW:    www.neuroinformatik.de

freyzor, 2015-02-19:
managed to boil it down to a pretty straight forward repro
```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
    <macro name="coord" params="xy">
        <point xy="${xy}" />
    </macro>

    <coord xy="5 2" />
    <coord xy="5 -2" />
    <point2 xy="5 -2" />
</robot>
```
This will return the following
```xml
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
  <point xy="5 2"/>
  <point xy="3"/>
  <point2 xy="5 -2"/>
</robot>
```

It appears that when passing in something that looks like an expression to the macro param it will evaluate it as such rather than forwarding it as is. I was under the impression that python would only evaluate with in the ${} brackets so is rather unexpected :)

rhaschke, 2015-02-19:
Thanks for pointing out this issue. Fortunately, there is an easy fix, which is available in the [latest version](https://github.com/ubi-agni/xacro/tree/python_eval).

freyzor, 2015-02-20:
*hi5* that was quick.  Many thanks. I updated and is working very well so far :)  I can happily close this issue now.

ps. the property macro I suggested does not work for a couple of unrelated reasons but I can well live with that.

closed 2015-02-20 by freyzor