ros/xacro #51: Ubuntu 12.04 DEB package: unnecessary dependencies. Subversion?

kamiccolo, 2015-02-23:
Recently while tinkering with some ROS builds my eye caught these kind
of lines:

$ dpkg -s ros-groovy-xacro
        Package: ros-groovy-xacro
        Depends: ros-groovy-ros (= 1.9.53-0precise-20140905-2256-+0000),
ros-groovy-ros-comm (= 1.9.55-0precise-20140906-0133-+0000), libc6,
build-essential, cmake, python-yaml, subversion.

After brief source-code review I
can't see any piece of code relevant or anyhow related to **subversion**. Is it really needed for *ros-groovy-xacro` runtime? The same applies to **cmake** and **build-essential**.

Operating System: Ubuntu 12.04 running stock ROS Groovy.

codebot, 2015-02-23:
So far as I understand, the subversion dependency is chaining in from the rosbuild dependency of ros-groovy-xacro, which (at that time) did not separate "build" from "run" dependencies. That's because groovy was using the "old" rosbuild system, instead of the "current" build system, called catkin, which separates out various kinds of dependencies. If you upgrade to hydro or indigo, hopefully the dependencies are significantly cleaner.

kamiccolo, 2015-02-23:
*sigh* Deprecated rosbuild leftovers makes sense. Thanks.

closed 2015-02-23 by wjwwood