ros/xacro #52: Usage of block parameter

130s, 2015-03-04:
I made [a change on urdf-xacro tutorial](http://wiki.ros.org/action/info/urdf/Tutorials/Using%20Xacro%20to%20Clean%20Up%20a%20URDF%20File?action=diff&rev2=13&rev1=12) about `block parameter`, but I'm not confident if this is right. Should it be wrong someone please correct that.

130s, 2015-03-04:
@DLu

DLu, 2015-03-04:
What was wrong with it in the first place? The idea was to show that you could plug in any shape, not just cylinders, to make a generic blue SHAPE.

130s, 2015-03-14:
> The idea was to show that you could plug in any shape, not just cylinders, to make a generic blue SHAPE. 

Yeah, I know plugging in specific shape like cylinders doesn't sound good. 

> What was wrong with it in the first place?

`shape` isn't defined anywhere in [the previous edition](http://wiki.ros.org/action/recall/urdf/Tutorials/Using%20Xacro%20to%20Clean%20Up%20a%20URDF%20File?action=recall&rev=12). In my understanding the block has to be defined somewhere. In [this example](https://github.com/ipa320/schunk_modular_robotics/blob/ed802939410d977cc220dec6106aaa0a699ae5f2/schunk_description/urdf/pg70/pg70.urdf.xacro#L12), `origin` block is defined many times later in the same file.

DLu, 2015-03-17:
The idea is that shape can represent any kind of block element. `shape` is just the name of the parameter. The macro takes whatever block is defined in the macro (`<cylinder radius=".42" length=".01" />`) and inserts it where it says `insert_block`.

DLu, 2015-03-17:
I have reverted the wiki change.

closed 2015-05-22 by codebot