ros/xacro #53: expat parsing error

sivammt, 2015-03-09:
I was trying to execute my urdf.xacro file and I get the following error. can anyone help me in explaining the details of this error.

[ INFO] [1425902453.451149717]: Running 'rosrun xacro xacro.py /home/siva/catkin_ws/src/arm_and_hand_motor_biotac.urdf.xacro'...
Expat parsing error.  Check that:
 - Your XML is correctly formed
 - You have the xacro xmlns declaration: xmlns:xacro="http://www.ros.org/wiki/xacro"

Traceback (most recent call last):
  File "/opt/ros/hydro/share/xacro/xacro.py", line 60, in <module>
    xacro.main()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/xacro/__init__.py", line 646, in main
    doc = parse(f)
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 1920, in parse
    return expatbuilder.parse(file)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 928, in parse
    result = builder.parseFile(file)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 207, in parseFile
    parser.Parse(buffer, 0)
xml.parsers.expat.ExpatError: junk after document element: line 28, column 0

codebot, 2015-03-09:
This error is a generic XML parser error; without seeing the .xacro input, it's hard for us to tell what's going on. Can you post the XML file(s) that cause this error?

sivammt, 2015-03-10:
I have already opened an issue regarding this and i got the reply. The reply is as follows

On the shadow hand everything is generated via macros. Please look at sr_description/robots/arm_and_hand_motor.urdf.xacro 
it is basically a description doing
world -> world to arm joint -> arm macro -> arm to hand joint ->hand macro

the way to go for you is then 
world -> world to kuka joint -> kuka -> kuka to shadow fixed joint -> hand macro.

of course this creates a xacro file that should be expanded with a xacro.py call (this is standard to load robot_description on the fly)

I proceeded with the above instructions and i changed the urdf. The changed urdf is as follows.

<robot xmlns:xacro="http://ros.org/wiki/xacro" name="arm_and_hand">

  <xacro:include filename="$(find sr_description)/materials.urdf.xacro" />
  <xacro:include filename="$(find sr_description)/arm/xacro/kuka.urdf.xacro" />
  <xacro:include filename="$(find sr_description)/hand/xacro/full_hand_motor_biotac.urdf.xacro" />
  <xacro:include filename="$(find sr_description)/other/xacro/gazebo/gazebo.urdf.xacro" />

  <link name="world" />
  <joint name="kuka_to_world_fixed" type="fixed">
    <parent link="world"/>
    <child link="kuka_link_base"/>
    <origin rpy="0 0 0" xyz="0 0 1.0"/>
  </joint>

  <xacro:sr_kuka>
  </xacro:sr_kuka>

  <joint name="arm_link" type="fixed">
    <parent link="kuka_link_7"/>
    <child link="forearm"/>
    <origin xyz="0 0 0" rpy="0 0 ${2.35619449}" />
  </joint>

  <xacro:shadowhand_motor>
  </xacro:shadowhand_motor>

</robot>
<?xml version="1.0" ?>

and i placed the kuka urdf file in the respective system folder. the kuka urdf is as follows 

<?xml version="1.0"?>
<robot name="kuka">

  
  
  

    <material name="IPA/Orange">
      <color rgba="1.0 0.5 0.0 1.0"/>
    </material>
    <material name="black">
      <color rgba="0.5 0.5 0.5 1"/>
    </material>
        
    <link name="kuka_link_footprint"/>

    <link name="kuka_link_base">
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry>
        <box size="0.01 0.01 0.01"/>
      </geometry>
      <material name="black">
      </material>
    </visual>
    </link>
[... 118 lines cut ...]
    <parent link="kuka_link_5"/>
    <child link="kuka_link_6"/>
  </joint>

  <link name="kuka_link_7">
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://kuka_barrett_description/mesh/joint_7.stl" />
      </geometry>
      <material name="IPA/Orange">
      </material>
    </visual>
  </link>
  
  <joint name="kuka_joint_7" type="revolute">
    <origin xyz="0 0.057 0.0455" rpy="-1.570796326 0 0" />
    <axis xyz="0 0 1" />
    <limit effort="30" velocity="1.9634" lower="-2.9670" upper="2.9670" />
    <parent link="kuka_link_6"/>
    <child link="kuka_link_7"/>
  </joint>
</robot>

kindly suggest me a solution.

sivammt, 2015-03-10:
I have linked the kuka arm and shadow hand successfully. I used kuka arm urdf in place of full_arm_motor.urdf . 
(<xacro:include filename="$(find sr_description)/arm/xacro/full_arm_motor.urdf.xacro" />)
but i forgot to shift the .stl files of KUKA into the respective folder. 
Now it works. thank you

closed 2015-03-10 by codebot