ros/xacro #57: increase speed by replacing xml.minidom with lxml.etree

codebot, 2015-03-16:
It appears that we can speed up xacro a lot by moving to lxml.etree instead of using xml.minidom. Obviously this would be a big change.

@mikepurvis , @paulbovbel , @mikeferguson , @wjwwood , and anyone else: any comments or feedback on this proposal?

Related: I'm also interested in adding some more large / complex "full-robot" test cases to xacro, to complement to the PR2 full-robot test case. I think we'd have more confidence when making big changes like this to xacro if we had more large test cases. Anyone have suggestions for which robots to add?

mikepurvis, 2015-03-16:
Flagging @TheDash, who is presently maintaining PR2.

Is the speed an issue for anyone? Our xacro is all extremely simple, and the compositing step only runs once on launch anyway.

If there are resources available on this, I would be interested in seeing a survey of the many, many other schemes which now exist for templating and compositing XML, and an assessment given of the cost of creating a ROS wrapper/plugin for one of them which gives the necessary path-resolution stuff but otherwise weakens the coupling to the ROS ecosystem. Something like Genshi or PyMeld, or even non-XML tools like Empy or Jinja2.

codebot, 2015-03-16:
@rhaschke has reported that some of their large files are taking up to 30 seconds.

TheDash, 2015-03-16:
Never had an issue with the xacro compiling. Not speed at least, but with other things (parameters, tests, structure and composite capability)

TheDash, 2015-03-16:
If needed, we can run Vtune on Xacro to see where some bottlenecks in the code are.

codebot, 2015-03-16:
@rhaschke  has done some profiling, which he summarized briefly in #56 , which is why he suggested we switch to a faster XML parser.

mikeferguson, 2015-03-16:
I've not really had an issue with speed -- our use of Xacro is pretty simple. Most of our robot is a single URDF (this is going to be more and more common as more people export their URDF directly from Solidworks) and then the gripper gets added via Xacro (since it is a modularity point). My other use of Xacro is to incorporate the tags required for Gazebo.

I can supply test Xacros in the future, but not at this time as the robot is not yet released.

wjwwood, 2015-03-16:
In principal the result should be the same, but there is a lot of opportunity to introduce new bugs. It could be possible to make it an option, depending on how complex the changes are. It might be a good opportunity to introduce an abstract internal to xacro, allowing it to use different xml engines.

Overall, I'm agnostic. I'm fine with users who need the performance being a champion for moving to a new xml parse, but the burden is on them and the maintainer of xacro to not break other users.

rhaschke, 2015-03-16:
We use xacro to assemble the robot system description in a modular fashion: we have several arms available, several hands as endeffectors and the hands (Shadow Hands) come in various different flavours. This all requires a lot of macros and includes which then sums up to long loading times - particularly on NFS volumes.
I will check whether we can provide our files as a test case.

rhaschke, 2015-03-16:
This is the result of profile.run on our setup:
Most time is spent in xml.minidom.insertBefore followed by QuickLexer, which reinstantiates the regular expressions over and over.
[image]

codebot, 2015-05-22:
#62 merged, which should speed things up considerably; closing this discussion for now. Feel free to re-open if further speed improvements are needed.

closed 2015-05-22 by codebot

rhaschke, 2015-06-10:
Fixing a performance issue in xml.dom.minidom, I was able to speedup our xacro calls by a factor of 10.
I filed a corresponding bug: https://bugs.python.org/issue24424