ros/xacro #61: Parsing Error for nested evaluations not really helpful

simonschmeisser, 2015-04-09:
Doing a nested evaluation like ${a/2+${b/2}} is not supported it seems. That's fine with me. However the error message could be more descriptive. Right now I get

    rosrun xacro xacro.py workcell.xacro > workcell.urdf 
    Traceback (most recent call last):
    File "/opt/ros/indigo/share/xacro/xacro.py", line 60, in <module>
    xacro.main()
      File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 692, in main    
    eval_self_contained(doc)
      File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 627, in eval_self_contained
    eval_all(doc.documentElement, macros, symbols)
      File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 552, in eval_all
    eval_all(body, macros, scoped)
      File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 610, in eval_all
    result = eval_text(at[1], symbols)
       File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 482, in eval_text
    results.append(handle_expr(lex.next()[1][2:-1]))
      File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 469, in handle_expr
        return eval_expr(lex, symbols)
      File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 450, in eval_expr
        n = eval_term(lex, symbols)
      File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 422, in eval_term
        n = eval_factor(lex, symbols)
      File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 392, in eval_factor
    if lex.peek()[1] == '-':
    TypeError: 'NoneType' object has no attribute '__getitem__'

rhaschke, 2015-04-23:
Nested evaluations as you did are not supported, but its even easier: ${a/2+b/2} should work!
In `jade-devel`, the error message slightly improved:
```
  a/2*${b/2
      ^
SyntaxError: invalid syntax
```

simonschmeisser, 2015-04-23:
ah nice, thats detailed enough for me

closed 2015-04-23 by simonschmeisser