ros/xacro #63: Don't evaluate substitution_args in conditional blocks with unsatisfied condition.

peci1, 2015-04-28:
Hi, I write a piece of code that assembles a robot from various parts. My idea is to allow users to only download some repositories, and using Xacro arguments to specify which parts should be assembled into the URDF.

Thus the current behavior of Xacro is quite unpleasant for me, because it requires the user to have all the model packages downloaded even if he doesn't want to use models from some of them.

Here's an example:
    <xacro:arg name="has_stuff" default="0" />
    <xacro:if value="$(arg has_stuff)">  
        <xacro:include filename="$(find nonexistent)/urdf/model.urdf" />
    </xacro:if>

The current implementation fails with not finding package `nonexistent`. 

Citing from wiki.ros.org/xacro:
  
    Additionally, the conditional has no effect on macro or property definitions, and is only used while constructing the final xml.

From this sentence, it's still not apparent to me if the behavior I address as a bug should be considered so. However, I think think supporting such things would be good for extending the possible use-cases for Xacro.

I know the fix won't be easy, but essentially I think most of it relates to the recently added support for `<xacro:arg>`. In my view, there should first be a run where all `<xacro:arg>`s would be evaluated, then the conditionals, and then the rest. However, I didn't have time to go through the Xacro source thoroughly, so there may be some catches I don't know - e.g. if there can be a property in the conditional test, then it'd need to be also evaluated before the conditionals...

rhaschke, 2015-05-07:
In order to realize your requested behavior, I implemented an in-order processing mode a while ago (60c958f5ff38f89956f55d37d0a5db945ff6a425). From my point of view, your suggested patch will further obscure the processing pipeline.
Due to backward compatibility constraints, the new in-order processing needs to be switched on using the cmdline argument --inorder.

peci1, 2015-05-07:
Hmm, good. Is there a possibility it'd be backported to indigo?

rhaschke, 2015-05-07:
I don't think so. The maintainers didn't wanted to merge these changes to indigo-devel in order to not break existing installations (because of the slightly different, but IMHO more natural parsing semantics).
However, xacro is rather small package that you easily add to your workspace without slowing down builds.

peci1, 2015-05-07:
Okay, I just hoped to be able to use the standard ROS Indigo parts packaged for Ubuntu, but including a custom version of Xacro looks also as an option. Thanks god for catkin :-D And thank you, too.

codebot, 2015-05-22:
As @rhaschke explained, our top priority with already-released ROS distributions is to not break existing functionality. Everyone understands that when upgrading from one distribution to the next, there may be (hopefully small) tweaks required, but I'm worried, perhaps overly so, about changing behavior of released distros. The official Jade release is imminent, where the --inorder feature is available. Thank you for your time and contributions.

closed 2015-05-22 by codebot