ros/xacro #100: auto-forward property definitions to macro parameters?

rhaschke, 2015-06-29:
I often encounter the situation, that I need to pass existing properties to identically named macro parameters (e.g. var="${var}"). I can circumvent that by not declaring these parameters. However, as soon as they are declared, xacro expects the parameters to be explicitly provided.

I suggest to relax this strong constraint and only complain if a property is not defined at all, i.e. neither explicitly in the macro params nor implicitly in the outer scope.
This will facilitate to explicitly _declare_ all used macro parameters while avoiding to explicitly _provide_ them.

What do you think?

rhaschke, 2015-06-29:
Kind of the same effect can be achieved by using default param values equal to existing properties:
```xml
<xacro:macro name="xacro:test" params="var:=${var}"/>
```
On the one hand, this explicitly states that the default is taken from the outer scope.
On the other hand, it prevents to define a real fallback default used when no outer scope property is defined. I suggest the following hierarchy:
* an explicitly provided argument takes highest priority
* if nothing is provided, but an outer-scope property with same name is defined, this one is (silently) taken
* otherwise, the default value (if defined) is used

If none of these cases apply, the warning about a missing argument is thrown.

guihomework, 2015-07-01:
> @guihomework
> I'm still not satisfied with the latest solution. It's too verbose. What about the following syntax:
> params="forwarded:=| forwarded_with_default:=|42"
> The pipe symbol (|) indicates either forward from outer scope or use the specified default.

The pipe symbol alone looks strange. In the minds pipe is or, which fits well to tell forward OR default
Is it possible to distinguish between ${} and $ alone ? because $ would actually make sense, you don't want to specify the name again, so $ is the place holder.
```params="forwarded:=$ forwarded_with_default:=$|42"``` 
Otherwise in some language, % means replacement for an argument too (in windows batch i think). 
```params="forwarded:=% forwarded_with_default:=%|42"```

rhaschke, 2015-07-01:
I prefer the `$|fallback` syntax. That's implemented on https://github.com/ubi-agni/xacro/tree/macro_args now (with all the other alternatives in the history).
@codebot: Please decide on a syntax proposed here and #101  ;-) 
I will then rebase and squash to have a clean pull request.

codebot, 2015-07-01:
could the python environment be modified to provide a "magic" symbol called "parent" or "super" that would resolve the symbols in the parent tag's scope, or return nil if the symbol doesn't exist? Then we could do things like this in the macro itself, rather than the argument list.
```xml
<a xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:property name="var" value="42"/>
    <xacro:macro name="foo">${parent.var or 42}</xacro:macro>
    <xacro:foo/>
</a>
```

rhaschke, 2015-07-01:
You can already (silently) use variables from the outer scope:
```xml
<a xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:property name="var" value="42"/>
    <xacro:macro name="foo">${var}</xacro:macro>
    <xacro:foo/>
</a>
```
However, I wanted to _explicitly_ declare all properties used in a macro. This has two benefits:
* macro code is more easily comprehensible (because more verbose/explicit)
* you can pass in other values to the macro without affecting the outside definition

The downside of an explicit parameter declaration, however, was that I was forced by xacro to explicitly provide the parameter, leading often to argument lists of the form
```xml
<xacro:foo arg1="${arg1}" arg2="${arg2}" arg3="${arg3}" arg4="${arg4}"  .../>
```

One could avoid that by providing default arguments like so:
```xml
<xacro:macro name="foo" params="arg1:=${arg1} arg2:=${arg2} arg3:=${arg3} arg4:=${arg4}">  ...
```

The drawback of this solution, is that
* it unnecessarily duplicates the var names and
* it doesn't allow to specify a fallback default anymore

Hence, I came up with this proposal:
```xml
<xacro:macro name="foo" params="arg1:=$|1 arg2:=$|2 arg3:=$|3 arg4:=$|4">  ...
```

To finally answer your question: Of course, it would be possible to introduce a parent keyword.
However, the variable - if it's not defined in the parent's scope - cannot be resolved anyway and trying to do so will result in an Exception:
```python
> a or 1
NameError: name 'a' is not defined
```

codebot, 2015-07-01:
yeah, we'd have to catch the NameError and do some magic to look it up in the right symbol table and properly handle when it wasn't there. That would probably be tricky. And reaching "up" to the parent scope is a bit gross for various reasons.

Anyway, thank you for explaining the issue again and providing more example snippets; I think I understand it better now.

In terms of a compact syntax for defining default argument values, I would suggest that we "borrow" a conditional-assignment operator from another language, rather than create a new one. For example:
 * GNU Makefile  ?=
 * Ruby  ||=  
 * PHP (I think?) uses ?: 
 * Switch (I think?) uses ??
 
Of all those choices, I guess I would lean towards the GNU Makefile option, but I don't have a hard preference.

For debugging / clarity, we could also perhaps add an attribute to xacro:macro to declare that only local properties can be used. Then, perhaps a simple example to exercise all of these features would be:

```xml
<a xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:macro name="foo" params="bar?=1 baz?=2" globals="false">${bar + baz}</xacro:macro>
    <xacro:foo bar="42"/>
</a>
```

and the desired evaluation result would be:
```xml
<a>44</a>
```

Does that look correct?

rhaschke, 2015-07-01:
This syntax doesn't yet cover the case of only forwarding the variable (without having a default fallback).
To summarize, by my current implementation the following variants will be provided:
```xml
<xacro:macro name="foo" params="no_default  forwarded_no_fallback:=$  forward_with_fallback:=$|1">
```
Going for makefile syntax `?=`, which syntax you would suggest for `forwarded_no_fallback`?
All your language examples are not really compatible to the current default syntax, starting with `:=`.

codebot, 2015-07-01:
Can you provide a complete example, including macro definition and invocation? I think I'm just not quite understanding it yet.

rhaschke, 2015-07-02:
Here is an example. Remember to use option `--inorder`, otherwise the last property definitions count.

```xml
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
	<xacro:macro name="foo" params="prefix:=$ name value:=$|42">
		<item name="${prefix}${name}" value="${value}"/>
	</xacro:macro>

	<!-- params 'prefix' and 'name' are required -->
	<xacro:foo prefix="" name="a"/>

	<!-- define a prefix default for the following items -->
	<xacro:property name="prefix" value="left_"/>
	<!-- only parameter 'name' is required now -->
	<xacro:foo name="a"/>
	<xacro:foo name="b"/>

	<!-- change defaults -->
	<xacro:property name="prefix" value="right_"/>
	<xacro:property name="value" value="21"/>
	<xacro:foo name="a"/>
	<xacro:foo name="b"/>

	<!-- having the parameters declared, we can also overwrite them ... -->
	<xacro:foo prefix="" name="c" value="17"/>

	<!-- ... and continue with the previous defaults -->
	<xacro:foo name="c"/>
</robot>
```

In my real-world example, I have a list of 10 parameters, that don't change very much. There, the saving in typing is even bigger.

codebot, 2015-07-02:
Thank you for writing out a complete example; I wasn't quite able to understand the goal from the previous snippets, but this idea makes sense to me now. I can see how this could dramatically reduce the number of characters required in a complex xacro file, which could improve readability and reduce maintenance pain.

My initial worry about using the dollar sign is that it's "overloaded" somewhat with all the other dollar signs that escape variable names and expressions, so it could be confusing for someone who inherits a xacro file to see the difference between ```params="prefix:=$ name``` and ```params="prefix:=$name"```  (the space between the dollar-sign and ```name``` makes a big syntactic difference).

Just to re-iterate, to make sure I've (finally!) understood the issue, there are four cases we want to handle nicely:

1. parameters with no default value
1. parameters with only a "classic" default value
1. parameters whose value can be taken from a parent-scope variable of the same name
1. parameters whose value can be taken from a parent-scope variable of the same name, or (if that doesn't exist) use a default value

Many languages use the double-colon for a scope-resolution operator to pull a symbol from the global namespace. That seems pretty relevant in this case. But it's a bit confusing to see single- versus double-colons, so one proposal would be to make the (plain) equals operator syntactically equivalent to the current := operator, and use the double-colon to mean "parent scope".

```xml
<xacro:macro name="foo" params="p1 p2=a p3=:: p4=::52>
```

The fourth case, however, is the tricky one, because the "typical" usage of double-colon followed by a value means to look up that value in the parent namespace. How about if we put the double-colon in front of the parameter name, instead of the default value? That might be kind of neat, because it's closer to the meaning of double-colons in other languages:

```xml
<xacro:macro name="foo" params="p1 p2=a ::p3 ::p4=b>
```

Of course, we'd need to still support the := syntax for "normal" default values for a very long time, possibly forever, until we call something "xacro2", because there are so many xacro files out there, but we could presumably support both = and := without difficulty in the parser; the "cleaner" use of the double-colon and equals sign here is just to make those operators align more closely with how they are usually employed for default-parameter assignment and scope-resolution in popular languages such as C++ and Ruby.

How does that look to you?

rhaschke, 2015-07-02:
> My initial worry about using the dollar sign is that it's "overloaded"
> somewhat with all the other dollar signs that escape variable names
> and expressions, so it could be confusing for someone who [...]
Actually the dollar sign tries to be a shortcut for `${name}`. Note that 
$name is not valid syntax for evaluation (actually it will considered 
plain text).
>
> Just to re-iterate, to make sure I've (finally!) understood the issue, 
> there are four cases we want to handle nicely:
>
>  1. parameters with no default value
>  2. parameters with only a "classic" default value
>  3. parameters whose value can be taken from a parent-scope variable
>     of the same name
>  4. parameters whose value can be taken from a parent-scope variable
>     of the same name, or (if that doesn't exist) use a default value
>
Exactly.
>
> Many languages use the double-colon for a scope-resolution operator to 
> pull a symbol from the global namespace. That seems pretty relevant in 
> this case. But it's a bit confusing to see single- versus 
> double-colons, so one proposal would be to make the (plain) equals 
> operator syntactically equivalent to the current := operator, and use 
> the double-colon to mean "parent scope".
>
> <xacro:macro  name="foo"  params="p1 p2=a p3=:: p4=::52>
>
> The fourth case, however, is the tricky one, because the "typical" 
> usage of double-colon followed by a value means to look up that value 
> in the parent namespace. How about if we put the double-colon in front 
> of the parameter name, instead of the default value? That might be 
> kind of neat, because it's closer to the meaning of double-colons in 
> other languages:
>
> <xacro:macro  name="foo"  params="p1 p2=a ::p3 ::p4=b>
>
> Of course, we'd need to still support the := syntax for "normal" 
> default values for a very long time, possibly forever, until we call 
> something "xacro2", because there are so many xacro files out there, 
> but we could presumably support both = and := without difficulty in 
> the parser; the "cleaner" use of the double-colon and equals sign here 
> is just to make those operators align more closely with how they are 
> usually employed for default-parameter assignment and scope-resolution 
> in popular languages such as C++ and Ruby.
>
I understand your motivation for the double colon. However, they 
somewhat deviate from existing xacro syntax. For me, they are less 
intuitive than my proposal

`<xacro:macro  name="foo"  params="p1 p2:=a p3:=$ p4:=$|b>`

This more explicitly states the intend: _use var ($) from outer scope or 
the fallback default_
Shall we collect some more opinions on that from the community?

codebot, 2015-07-02:
yeah, it's certainly a good idea to ask for other opinions.

"what if" we put the dollar sign in front of the variable name, to try to avoid needing an "or" operator for case (4)?

```xml
<xacro:macro name="foo" params="p1 p2:=a $p3 $p4:=b">
```

Then, the only "new magic" we are adding is that declaring a parameter with a dollar-sign in front `$param` is shorthand for `param:=${param}` for case (3), plus an "additional" default value, with the "original" xacro default-value syntax for case (4).

How does that look to you?

rhaschke, 2015-07-02:
I would prefer a syntax that _appends_ stuff after the `:=` operator that currently indicates a default value.
Of course your proposals of prepending a double colon `::` or the dollar sign `$` in front are feasible as well. IMHO they are less intuitive - you cannot grasp the semantic from a first look. Of course, I'm also not sure that my proposal is intuitive - as I circled around the topic for a while already, I might have become blind.

codebot, 2015-07-06:
Yeah, this is a tricky one. Let's keep brainstorming. I'm sure we'll find something reasonable. I'm warming up to your suggested syntax, since I think my latest proposal is borderline incomprehensible.

I'd still like to base it off some pre-existing idiom in some major programming language, but perhaps that's hard to find. In Ruby you can use the double-bar operator || as a logical-OR in expressions where the type "falls through," so doing this:

```
foo(bar || 42)
```

will evaluate as `foo(42)`  if `bar` is null or false. But, like Python, you'd get an undefined-variable exception if ${var} was undefined, which is the "typical" case we are trying to shorten here. I can now understand why your original proposal in #101 was for a new function, `forward()`, which tested for existence of the property and, if it didn't exist, used the (optional) default value. That's how you'd have to implement it in a "major" programming language, to my knowledge, at least.

What other operators could we use than $ and |  ? I fear the dollar-sign is becoming "overloaded" a bit too much. What if we used asterisk, * instead? This would be an attempt at replicating the pointer-dereference operator in C/C++. If we combined this with the "logical-OR" operator , we'd arrive at a proposal like this:

```xml
<xacro:macro name="foo" params="p1 p2:=a p3:=* p4:=*||b">
```

Of course, that has its own problems, such as the use of asterisk in shell-glob operations. Let's keep brainstorming. What other operators / symbols can we use for the shorthand for expressions of the form `bar:=${bar}`   ?

paulbovbel, 2015-07-06:
Hello all,

Just to double check, the discussion is on how to reduce the following syntax:

```xml

<xacro:macro name="foo" params="p1 p2:=1">
</xacro:macro>

<xacro:foo p1="1" p2="${p2}"/>

```

...to get rid of the 'p2' call in the macro invocation.

I would like to express my feeling as a user, that adding a new unfamiliar operator shorthand, and a layer of indirection through forwarding, may be solving a problem that really isn't significant. At the point that a macro is receiving an inordinate amount of information from the parent scope, it's probably time to refactor into a more encapsulated structure.

codebot, 2015-07-06:
Hi Paul,

Thanks for the input; it is greatly appreciated. Having more voices in the discussion always helps!

I think that @rhaschke 's use case involves lots of parameters, where the shorthand could provide a significant time and maintenance savings. @rhaschke , can you share the xacro files that are providing the motivating use for this feature? Perhaps others could then provide some ideas and suggestions, as well as understand better the motivating application for this feature.

Cheers

paulbovbel, 2015-07-06:
Thanks @codebot, a more demonstrative implementation would be very helpful. I'm very thankful for the work that's been put into xacro for jade, and I'm very excited to take advantage of the new developments in improving our URDFs for jade.

rhaschke, 2015-07-07:
@paulbovbel 
Indeed, I have a lot of parameters to pass through several macros and it heavily facilitates writing if I could define those properties _once_ at some global level. Of course, I could to so and simply not declare the macro parameters at all (but simply used the global properties silently). However, I like to have all parameters explicitly declared in order to know what's going on... Furthermore, having macro parameters defined, allows to (locally) redefined them if necessary.

I have a nice example for defining tactile markers in the robot_description for the Shadow Robot Hand:
There are three files involved:
* [full_hand.tactile.xacro](https://github.com/ubi-agni/sr-ros-interface/blob/ubi-indigo-devel/sr_description/hand/xacro/full_hand.tactile.xacro) defining the top-level macro `shadowhand_tactile`
* [tactile.ubi.xacro](https://github.com/ubi-agni/sr-ros-interface/blob/ubi-indigo-devel/sr_description/hand/xacro/tactile.ubi.xacro) defining macros for individual taxel groups: 
`finger_tactile`, `thumb_tactile`, and `palm_tactile`, which in turn use macros `proximal_tactile`, `middle_tactile`, `distal_tactile`.
* and finally [taxel.xacro](https://github.com/ubi-agni/sr-ros-interface/blob/ubi-indigo-devel/sr_description/hand/xacro/taxel.xacro) defining the actual XML-generating macro

Hence, I have a deep hierarchy of macros and I need to pass a lot of arguments. Having the automatic property passing in place, dramatically reduces the writing effort.

P.S.: Additionally to the xacro files, there are some yaml files defining the taxel properties (location and normal direction). They are located in `fingers/[distal|middle|proximal]/*.taxels.ubi.yaml`.

codebot, 2015-07-13:
@rhaschke thank you for linking to those xacro files; they are excellent examples of the proposal in question, as well as many of the other improvements you have recently done for xacro. That is an amazingly compact representation of such an intricate hand model!

I don't think that any of my proposals thus far have been improvements on the proposed syntax. I still worry a bit about the "dangling" dollar sign at the end of the assignment, since everywhere else in xacro, there is something "important" that follows the dollar sign. How about if we use dollar-sign plus a caret to signify "go up a scope" to look up the parameter value:

```xml
<xacro:macro name="foo" params="p1 p2:=a p3:=$^ p4:=$^|b">
```

I'd also be in favor of (optionally) removing the colon from the colon-equals operator; it seems redundant and unnecessarily-different from C++, Python and other languages with default parameter values. But perhaps it's best to have that on a separate issue. And of course we should support the "old" := operator for the foreseeable future.

Anyway, @rhaschke what do you think of using `$^` to signify "look up in parent scope", and keeping the single-bar "OR" operator as proposed?

guihomework, 2015-07-14:
@codebot and @rhaschke I also like the up sign after (or maybe even without) the dollar sign
```xml
<xacro:macro name="foo" params="p1 p2:=a p3:=^ p4:=^|b">
```

rhaschke, 2015-07-14:
I will always advocate a simple syntax. Hence, I would prefer a single-char solution, like either the dollar sign or the caret (as suggested by @guihomework) but not both. 
Actually, I don't share your worries about the dangling dollar sign. For me, it's simply a shortcut for `${var}`, i.e. using the dollar indicates that I want to refer to some variable. Using only the caret, introduces a completely new syntax element, whose semantics doesn't become clear intuitively as well. 

What would be a good strategy to involve more xacro usesrs from the community in the discussion?
Could/should we have an online vote about the existing proposals?

Regarding your proposal to replace `:=` by a simple `=`, I agree in principle. However, this will complicate parsing dramatically. Currently `:=` is quite unique and we can easily split the parameter string at this string. If we move to a simple equal sign, we cannot do so anymore, because an equal sign can naturally show up in the default expression as well, e.g. `${var='foo'}`. Hence, we would need to explicitly parse these expressions... Maybe we should aim for that anyway as currently it's forbidden to have spaces in these expressions (spaces are used to split individual params). However, I didn't have a simple idea yet to parse these expressions, as we would need to deal with quoting etc.

codebot, 2015-07-14:
Thanks @guihomework and @rhaschke for the continued discussion. The single-caret sounds good to me. Even though it is a new syntax element, I think it is abbreviating something sufficiently complicated that it's good to introduce a new element to clarify it. Users will have to read the docs about this feature, no matter what we do.

Sorry to have introduced the equals-sign simplification into this thread; it's a completely orthogonal issue that belongs in a separate issue (someday). As @guihomework  suggested, the proposed syntax is now:

```xml
<xacro:macro name="foo" params="p1 p2:=a p3:=^ p4:=^|b">
```

I like this, and would be happy to merge it anytime. Thanks @rhaschke  for all of your work on this feature.

In terms of reaching more people, I'm not sure how to do that; we've been having this discussion in the open already, and those who are watching this repo can always join in whenever they wish.

rhaschke, 2015-07-15:
Good, if Guillaume's syntax has a consensus, I will adapt the parsing code appropriately and file a push request.

codebot, 2015-07-24:
closing this issue now, since #105 is now merged. Thanks everyone for the input and @rhaschke for the PR.

closed 2015-07-24 by codebot