ros/xacro #107: Feature: Second interface so that functionality can be used in own script

NikolasE, 2015-08-06:
I'd so far seen no way to run xacro without using the main()-function that reads sys.argv. I'd however like to convert a urdf.xacro to an urdf within my own python script. 

I am missing this functionality when I create the urdf model during the startup-process of my robots. 
I store some xacro-arguments in a database so that all robots have the same code-basis but the correct urdf is created. As the xacro/__init__py.main() assumes to be called as it's own program (using sys.argv) it's hard to include the urdf-creation can be included in my own process while I still have access to all Exceptions (e.g. if an argument was not set). 

The interface would look something like

# urdf_xacro_filename: absolute path to robot.urdf.xacro
# mappings: a dict of xacro-arguments
def convert_urdf(urdf_xacro_filename, mappings):
     (....)
     return doc.toprettyxml(indent='  ')

Exceptions should be re-raised so that the user of this function could handle them (e.g. just output on rospy.logginfo()). 

What do you think about such in interface? I'd be happy to implement it, if there's a good chance that it would be integrated.

rhaschke, 2015-08-06:
The functionality you request is available from a very few function calls:
```python
opts = xacro.process_args(arg_string)
doc = xacro.parse(filename)
xacro.process_doc(doc, **vars(opts))
return doc.toprettyxml(indent='  ')
```
The essential functionality provided in xacro.main() is to handle the exceptions nicely.
As you write, you want to handle exceptions in your fashion. Hence you should kind of reimplement xacro.main() with your own exception handling.

When refactoring xacro for jade, I already tried to have as few high-level functions as possible and the above set of functions is the result. As all calls except `process_doc()` might be optional in various settings, I don't see the need for another function.

codebot, 2015-08-10:
@NikolasE  , does the functionality described by @rhaschke meet your needs?

closed 2015-12-04 by codebot