ros/xacro #108: BC break: expressions and macros inside property-block definition are not expanded

peci1, 2015-08-12:
<root version="1.0" xmlns:xacro="http://www.ros.org/wiki/xacro">
            <xacro:macro name="test">
                    ahoj
            </xacro:macro>
            <xacro:property name="val" value="2" />
            <xacro:property name="prop">
                    <outer attr="${val}">
                    testy
                    <xacro:test />
                    </outer>
            </xacro:property>
            <xacro:macro name="final">
                    <xacro:insert_block name="prop" />
            </xacro:macro>
            <xacro:final />
    </root>

This snippet generates the following in Indigo:

    <?xml version="1.0" ?>
    
    
    
    
    <root version="1.0" xmlns:xacro="http://www.ros.org/wiki/xacro">
      <outer attr="2">
    
                    testy
    
    
                    ahoj
    
      </outer>
    </root>

And the following in Jade (regardless of --inorder):

    <?xml version="1.0" ?>
    
    
    
    
    <root version="1.0" xmlns:xacro="http://www.ros.org/wiki/xacro">
      <outer attr="${val}">
    
                    testy
    
        <xacro:test/>
      </outer>
    </root>

rhaschke, 2015-08-12:
@codebot: I will have a look.
Best, Robert

rhaschke, 2015-08-12:
Fixed with #109.

commit 3bd26a7b (2015-08-12): Merge pull request #109 from ubi-agni/fix#108

fix #108: evaluate property blocks recursively too
commit c938b6d9 (2015-08-12): fix #108: evaluate property blocks recursively too
closed 2015-08-12 by codebot