ros/xacro #115: Can't generate launch files with xacro include

trainman419, 2015-09-10:
xacro's old include syntax collides with roslaunch's include syntax, making it difficult to generate launch files using xacro.

When trying to generate launch files using xacro, I get:

```
DEPRECATED IN HYDRO:
  The <include> tag should be prepended with 'xacro' if that is the intended use
  of it, such as <xacro:include ...>. Use the following script to fix incorrect
  xacro includes:
     sed -i 's/<include/<xacro:include/g' `find . -iname *.xacro`
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/xacro/xacro", line 53, in <module>
    xacro.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 687, in main
    process_includes(doc, os.path.dirname(args[0]))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 271, in process_includes
    raise XacroException("included file \"%s\" could not be opened: %s" % (filename, str(e)))
xacro.XacroException: included file "" could not be opened: [Errno 2] No such file or directory: ''
```

Since xacro now has the `<xacro:include>` tag, would it make sense to add a flag to xacro which disables the deprecated include syntax?

rhaschke, 2015-09-10:
In xacro-jade there is already an option `--xacro-ns` that enforces usage of the `xacro:` namespace prefix. However, unfortunately, this didn't yet applied for `<include>` tags. Fixed now.

rhaschke, 2015-09-10:
This option was already documented: http://wiki.ros.org/xacro#Deprecated_Syntax

commit 22ce42de (2015-09-10): Merge pull request #116 from ubi-agni/jade-devel

fix #115: enforce xacro namespace usage with --xacro-ns option
commit 534675a5 (2015-09-10): fix #115: enforce xacro namespace usage with --xacro-ns option

apply correct checking for include tags too
extended the testcase to include
closed 2015-09-10 by codebot