ros/xacro #118: enforce xacro namespace for created macros if --xacro-ns flag is set

ipa-mdl, 2015-11-27:
Calling `rosrun xacro xacro --inorder --xacro-ns test.xacro` with the following `test.xacro`
```xml
<test xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:macro name="printer" params="text" >
        <print what="${text}"/>
    </xacro:macro>
    <printer text="hello world"/>
</test>
```
results in
```xml
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from test.xacro                     | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<test xmlns:xacro="http://www.ros.org/wiki/xacro">
  <print what="hello world"/>
</test>
```

Is this intended?

This is especially a problem if you use the macro name inside the macro definition, e.g.:
```xml
<test xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:macro name="printer" params="text" >
        <printer what="${text}"/>
    </xacro:macro>
    <printer text="hello world"/>
</test
```
which results in

```
Invalid parameter "what"
when instantiating macro: printer (test.xacro)
instantiated from: printer (test.xacro)
in file: test.xacro
```
This issue is related to #115.

rhaschke, 2015-11-27:
This is kind of intended behaviour. I remember that we discussed that issue and decided that the user is free to choose a different macro name.
What behaviour do you expect? `--xacro-ns` enforcing the `xacro:` namespace for macro names too?
@codebot I will prepare a PR to realize that behaviour.

rhaschke, 2015-11-28:
Looking into the code, I found that we have implemented another way to enforce the `xacro:` namespace usage on macro usage: If the macro is defined with the `xacro:` namespace in front, you are also enforced to use the prefix on usage:

```xml
<test xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:macro name="xacro:printer" params="text" >
        <print what="${text}"/>
    </xacro:macro>
    <printer text="hello world"/>
    <xacro:printer text="hello world"/>
</test>
```

results in:
```xml
<test xmlns:xacro="http://www.ros.org/wiki/xacro">
  <printer text="hello world"/>
  <print what="hello world"/>
</test>
```

Probably, we should better document this ;-)

ipa-mdl, 2015-11-28:
> What behaviour do you expect? --xacro-ns enforcing the xacro: namespace for macro names too?

Yes, that's what I expected..

>  If the macro is defined with the xacro: namespace in front, you are also enforced to use the prefix on usage

I have already found that out, but it felt rather hackish ^^
Some legacy xacros still do not use the `xacro:` namespace and I had to find a solution without `--xacro-ns`

> This is kind of intended behaviour. 

If this is intended then the documentation (especially the help text) should be improved.
Something like "require xacro namespace prefix for xacro tags, except user-defined macros"?

rhaschke, 2015-11-28:
I agree that the current behavior is somewhat awkward. If there is a consensus to switch to the proposed behavior, there is a PR #119 doing the job.

closed 2015-11-30 by codebot

ipa-mdl, 2015-11-30:
Great, thanks! :)

PR #119 (2015-11-30): always store macros with xacro: prefix in front (fix for #118)
This PR stores all parsed macros with the `xacro:` prefix in front.
At usage, the `xacro:` prefix is expected by default too. However, there is a fallback (if `--xacro-ns` wasn't specified) to also find the macro when the prefix wasn't provided. This reverses the previous processing order, which first tried to resolve for the macro name without and then with the prefix.
This resolves #118 providing more consistent behavior.