ros/xacro #128: Empty default value for args causes ArgException

T045T, 2015-12-21:
The following xacro, when invoked with ```rosrun xacro xacro.py test.xacro```, breaks with ```roslaunch.substitution_args.ArgException: foo```
Seems like empty defaults are not treated properly (the URDF would be valid even with the default value of "empty string")
```
<?xml version="1.0"?>
<robot name="bar"
       xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:arg name="prefix" default=""/>
  <link name="$(arg foo)link" />>
</robot>
```

The equivalent launch file
```
<?xml version="1.0"?>
<launch>
  <arg name="prefix" default="" />
  <node pkg="tf"
        type="static_transform_publisher"
        name="$(arg prefix)pub"
        args="0 0 0 0 0 0 map base_link 100" />
</launch>
```
works as expected.

I can try and look into this myself, but won't get to it for a while (I've used a workaround for the original use case). If one of xacro's maintainers already knows where to make a fix, that would probably be quicker :)

rhaschke, 2015-12-21:
Your xml snippet is probably wrong. `foo` should read `prefix` to match the argument name.
However, even then, the Indigo version of xacro fails with the error, because the code cannot yet distinguish between no default value and an empty string.
This issue was fixed in the `jade-devel` branch of xacro, which is also available in standard indigo distros using the `--inorder` commandline option.

T045T, 2015-12-21:
Oh, sorry, that's an artifact of me changing the xacro to be valid with the empty default value - previously I just had ```<link name="$(arg foo)" />```, which would have been invalid with an empty string :)

Anyway, thanks for the quick answer! I'll try and see whether ```--inorder``` breaks anything for our use case.

ipa-mdl, 2017-05-10:
Just stumbled upon this issue..

If using `--inorder` is not an option, this workaround can be used:
```
<?xml version="1.0" ?>
<robot name="test" xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:property name="prefix_default" value=""/>
  <xacro:arg name="prefix" default="${prefix_default}"/>

  <link name="$(arg prefix)link" />
</robot>
```

Panda1638, 2017-08-02:
To add something to the general topic: you can't use underscores in front of arguments. Declaring _prefix was useless and ignored. After renaming the argument to prefix it worked.

rhaschke, 2018-03-28:
Fixed via #129.

commit 516d5af0 (2015-12-21): Merge pull request #129 from ubi-agni/indigo-devel

fix #128: allow for empty arg defaults
closed 2018-03-28 by rhaschke