ros/xacro #133: xacro:macro name

servetcoskun, 2016-06-03:
I think the above is broken again. In the kinetic-dev branch I'm getting the error:

`unknown macro name: xacro:base`

This is after running the fix script. In the xacro file it looks like this:
`<xacro:macro name>
...
</xacro:macro>
`

Just needs a 'replace' after line 499 in [__init__.py](https://github.com/ros/xacro/blob/kinetic-devel/src/xacro/__init__.py)??

Or maybe I just messed up my xacro file..

rhaschke, 2016-06-03:
Could you send your xacro file before and after auto-fixing?

servetcoskun, 2016-06-03:
basically [seven_dof_arm.xacro](https://github.com/qboticslabs/mastering_ros/blob/master/chapter_2_codes/mastering_ros_robot_description_pkg/urdf/seven_dof_arm.xacro) I've changed the original but the idea is the same.

rhaschke, 2016-06-04:
I would consider this a bug in your xacro file. The macros 
```
  <xacro:base name="base"/>
  <xacro:arm parent="base"/>
  <xacro:gripper parent="tool"/>
```
are never defined, only used. 
With old xacro, they show up literally in the final urdf, which I consider a bug too.

servetcoskun, 2016-06-05:
You are right, everything runs perfectly now. I will notify the repo owner asap, thanks for the help and sorry about the false alarm :)

closed 2016-06-05 by servetcoskun

pastoradrianv92, 2018-11-06:
> You are right, everything runs perfectly now. I will notify the repo owner asap, thanks for the help and sorry about the false alarm :)

HI THERE! PLEASE SHOW ME HOW DID YOU FIX IT, How did you declare? how did you change the seven_dof_arm.xacro?

wonggong, 2020-05-12:
> > You are right, everything runs perfectly now. I will notify the repo owner asap, thanks for the help and sorry about the false alarm :)
> 
> HI THERE! PLEASE SHOW ME HOW DID YOU FIX IT, How did you declare? how did you change the seven_dof_arm.xacro?

delete 
 <xacro:base name="base"/>
  <xacro:arm parent="base"/>
  <xacro:gripper parent="tool"/>