ros/xacro #134: Regression in xml include handling

cdellin, 2016-06-21:
I encountered a `xacro` error when attempting to build my robot model ([`herb_description`](https://github.com/personalrobotics/herb_description)) with ROS Kinetic.  I reduced it to the following minimal example:

```
$ cat a.xml
<root xmlns:xacro="http://www.ros.org/wiki/xacro">
<xacro:include filename="b.xml" />
</root>

$ cat b.xml
<root xmlns:xacro="http://www.ros.org/wiki/xacro">
<xacro:macro name="b">
   <element />
</xacro:macro></root>
```
(note the lack of newline in `b.xml` between `</xacro:macro>` and `</root>`)

On ROS Kinetic, I get the following error:
```
$ rosrun xacro xacro.py --verbosity=4 a.xml
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order

when processing file: a.xml

Traceback (most recent call last):
  File "/opt/ros/kinetic/share/xacro/xacro.py", line 61, in <module>
    xacro.main()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 1031, in main
    doc = process_file(input_file_name, **vars(opts))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 1008, in process_file
    process_doc(doc, **kwargs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 952, in process_doc
    grab_macros(doc, macros)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 529, in grab_macros
    grab_macros(elt, macros)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 527, in grab_macros
    grab_macro(elt, macros)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 490, in grab_macro
    remove_previous_comments(elt)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 785, in remove_previous_comments
    node.parentNode.insertBefore(_empty_text_node, next)
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 97, in insertBefore
    raise xml.dom.NotFoundErr()
xml.dom.NotFoundErr
```

Adding that newline to `b.xml` is a workaround.  This example parses correctly with or without the newline on ROS Indigo.  (I haven't tested on ROS Jade.)

rhaschke, 2016-06-22:
I will have a look.

rhaschke, 2016-06-22:
As a first fix, consider to use the new `--inorder` processing as suggested. This works like a charm on your example. Does it also work on your full file?

rhaschke, 2016-06-22:
I have pushed PR #135, which explicitly considers a corner case in upstream xml.dom.minidom.

codebot, 2016-06-22:
#135 fixes this for Jade; I'll see if I have enough git XP to pull that change into kinetic-devel as well.

codebot, 2016-06-22:
#136 fixes this for Kinetic. Thanks @cdellin for the excellent minimal issue report and @rhaschke for the fix. I'll push out releases right now.

closed 2016-06-22 by codebot

cdellin, 2016-06-22:
Thanks @rhaschke and @codebot for the quick fix!

commit 38900bfc (2016-06-22): Merge pull request #135 from ubi-agni/jade-devel

workaround for xml.dom.minidom issue (fixes #134)