ros/xacro #137: Unable to build package in Kinetic

chrisspen, 2016-07-02:
Attempting to build the package from source in Kinetic encounters an error:

    cd ~/ros_catkin_ws/src
    git clone https://github.com/ros/xacro.git
    cd ..
    catkin_make --pkg xarco
    Base path: /home/pi/ros_catkin_ws
    Source space: /home/pi/ros_catkin_ws/src
    Build space: /home/pi/ros_catkin_ws/build
    Devel space: /home/pi/ros_catkin_ws/devel
    Install space: /home/pi/ros_catkin_ws/install
    Packages "xarco" not found in the workspace

I've used this procedure to build many other packages without issue. What is the correct way to build xacro in Kinetic using catkin?

My platform is Raspbian running on a Raspberry Pi 3, where no pre-built packages are available.

codebot, 2016-07-02:
I think it just might be a typo; catkin is throwing an error because the
package requested is "xacro" instead of "xacro" . Hopefully using "xacro"
will work. Cheers

codebot, 2016-07-02:
sorry my phone autocorrected too much. The error message is for "xarco" but
I think you meant "xacro" 😁

chrisspen, 2016-07-02:
Indeed I did. Unfortunately, I wrote that on my laptop, so I have only myself to blame. Thanks.

closed 2016-07-02 by chrisspen