ros/xacro #141: Xacro $(find pkg) doesn't work for devel space assets

mikepurvis, 2016-08-10:
Roslaunch makes this work by checking for what follows the $(find) invocation, checking the devel space and falling back to the source space. This functionality is marked as being [for backward compatibility](https://github.com/ros/ros_comm/blob/98aa98e565ea623aae61c02f54ecb06c7ee190dd/tools/roslaunch/src/roslaunch/substitution_args.py#L117-L156), and doesn't seem to be present in xacro, although xacro calls into the same `substitution_args.resolve_args` function.

Looks like the issue is that Xacro treats find/arg/etc as "extensions" of its built-in expression processing capability, meaning that it's not trivial for it to pass the entirety of the attribute string to roslaunch. 

This limits the ability to generate at build time a URDF that is pulled into a runtime xacro, for example.

@dirk-thomas Thoughts about whether there's an easy way to fix this?

rhaschke, 2016-08-10:
To be honest, I found it always confusing that `$(find pkg)` isn't self-contained in `roslaunch`, but considers the following text too (as @mikepurvis pointed out, this is for backwards compatibility only). Hence, I would hesitate to introduce that feature in `xacro`.

mikepurvis, 2016-08-10:
It's _labeled_ as being for backward compatibility, but it's unclear to me how else you would get the required functionality. The reality of the devel space is that each package has two locations for resources— the path to the package.xml (in the source space) and the path to anything created at build time (in devel/share/<pkg>).

dirk-thomas, 2016-08-10:
I don't know enough about the internals of `xacro` to provide an answer.

`find` in `roslaunch` needs to looks a the whole argument. I agree that it is pretty ugly that the command in parenthesis is not self contained but needs to look ahead and consider what is coming afterwards. More than three years ago I proposed a cleaner approach for this (ros/ros_comm#228) but that was not accepted since no consensus could be reached in the discussion.

mikepurvis, 2016-08-10:
The relevant source is here:

https://github.com/ros/xacro/blob/23a0942b767823aec7f2ac50478e4e80d5903870/src/xacro/__init__.py#L597

I don't think it'd be too bad to change a few lines down where it gets called— you just pass the whole contents in, rather than the specific tokens.

rhaschke, 2016-08-14:
Making the extension evaluation not self-contained, but looking ahead past the parentheses, has several drawbacks:
- It reduces comprehensibility. 
- It's not clear where to stop reading ahead.
- Feeding back the information, where the file name actually ended, is not trivial as well.
- What if the file name itself is constructed from expressions?

I think, the clear solution is indeed something along https://github.com/ros/ros_comm/pull/228.

de-vri-es, 2016-10-03:
Where possible I prefer `package://` URIs resolved at runtime over `$(find pkg)`. The biggest advantage is that it doesn't break down when running nodes distributed over multiple machines with non-identical ROS package paths.

It also removes the need to do weird lookaheads outside of the arguments of `$(find)` since the URL resolver has access to the full URL. So maybe it would make more sense to add support for `package://` URLs where missing (such as the xacro include tag).

One downside I can see is that it means some `package://` URIs are resolved by xacro at expand-time while other are resolved by the consumer of the XML later. To me that seems acceptable though.

rhaschke, 2016-10-06:
I don't think that using `package://` URIs will solve @mikepurvis's issue: Those URIs will act in a similar fashion as a plain `$(find pkg)`, i.e. resolving to some fixed package path (`share`). However, Mike was asking for dynamic resolution, depending on the content after `$(find pkg)`.

de-vri-es, 2016-10-06:
Well, the URL resolver has access to the full URL including the bit after the package name, so searching multiple paths to find the resource is relatively straightforward. The `$(find pkg)` function only gets a package name or needs arguably strange lookahead behaviour. I do think that is an important difference.

rhaschke, 2018-03-28:
Closing this issue due to long inactivity. The last proposed solution seems feasible and clean.

closed 2018-03-28 by rhaschke