ros/xacro #143: Optional/default block parameters

dhood, 2016-09-06:
Is there anyway to specify block parameters as optional? Or specify the default value of block parameters?

Here's an example:
```
<?xml version="1.0"?>

<a xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:macro name="foo" params="*x">
    <xacro:insert_block name="x" />
  </xacro:macro>

  <xacro:foo>
    <x>4.5</x>
  </xacro:foo>
</a>
```

If I want to call the macro without specifying x, the only way I've found is to call it with an empty x block (otherwise I get `Not enough blocks while evaluating macro` exception). I'd prefer to not have the empty tag in the generated file, so thought I'd check if there's an alternative.

dhood, 2016-09-08:
This is what I have ended up using to prevent the empty x tag from being generated in the output:

```
<?xml version="1.0"?>

<a xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:macro name="foo" params="**y">
    <foo_data>
      <xacro:insert_block name="y" />
    </foo_data>
  </xacro:macro>

  <xacro:foo>
    <y><x>4.5</x></y>
  </xacro:foo>

  <xacro:foo>
    <y></y>
  </xacro:foo>
</a>

```

Which will generate:
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from /home/osboxes/xacro-test.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<a xmlns:xacro="http://www.ros.org/wiki/xacro">
  <foo_data>
    <x>4.5</x>
  </foo_data>
  <foo_data>
  </foo_data>
</a>
```

Other suggestions are still welcome if there's a better approach.

rhaschke, 2016-09-14:
`insert_block` inserts the _content_ of the block argument. Hence, you can specify an empty block argument to insert nothing. However, you cannot simply drop the block completely. Otherwise, if there are several block arguments, their order isn't well defined anymore...

dhood, 2016-09-14:
OK, and it doesn't sound like setting default values is possible. Thanks for the response!

closed 2016-09-14 by dhood