ros/xacro #144: Conversion from .xacro -> .urdf does not update xmlns correctly

dljsjr, 2017-01-24:
When a .xacro is converted to .urdf, the xmlns attribute on the robot tag is preserved as `xmlns:xacro="http://www.ros.org/wiki/xacro`

However, the URDF .xsd schema defines a `targetNamespace` attribute of http://www.ros.org. That means that a URDF's root robot tag should have `xmlns="http://www.ros.org"` and not `xmlns:xacro="http://www.ros.org/wiki/xacro`.

It's a small detail, however there exist many code generators/unmarshallers/validators/etc. that are very strict and will error out on this completely as if the generated XML document is malformed. An example of this is the JAXB tooling, which is built in to Java. If you use JAXB to generate Java bindings from the urdf.xsd and then try to use a JAXB unmarshaller on a .urdf with this bad xmlns then the unmarshalling process will hard fail and throw an exception.

codebot, 2017-01-27:
Thanks for pointing this out. It would be good to add a JAXB test to the xacro test suite. Can you point to a minimalist example of using JAXB which falls down with the current version of xacro? Thanks :chicken:

dljsjr, 2017-01-27:
@codebot I can probably fork this repo and add a self-contained test to the `test` directory some time next week.

codebot, 2017-01-27:
Fantastic! Thanks. :chicken:

rhaschke, 2018-03-28:
@dljsjr Do you still plan to provide a unittest?

dljsjr, 2018-03-28:
This totally fell off my radar when I moved on to another project like, the week after I filed this issue 😕

I have some free time in mid April where I can come back to this.

rhaschke, 2018-11-01:
@dljsjr I guess, that's not important to you anymore? Can we close the issue?
As xacro can be used for any XML and not only URDF creation, we would need a targetNamespace attribute in the root tag of the xacro document.

dljsjr, 2018-11-01:
@rhaschke Hey sorry! I still haven't transitioned back on to this, other projects pulled me away. It's still something we're interested in to be able to leverage more existing generic XML tools but we probably won't get back to it for a while, so if you need to close that's fine.

rhaschke, 2019-09-28:
Addressed in #207.

closed 2019-09-28 by rhaschke