ros/xacro #145: [xacro indigo] True/False vs. true/false when using conditional blocks with --inorder

ipa-fxm, 2017-01-30:
When testing the new possibilities that come with using the `--inorder` argument with xacro under indigo, we found the following behaviour - this possibly is a regression:
When using xacro:properties or macro parameters within a conditional expression e.g using `or`, the result is not as expected when the values of the properties are set to `true`/`false`. The result is correct when using `True`/`False`.
In [xacro](http://wiki.ros.org/action/fullsearch/xacro#Conditional_Blocks) the documentation says:
 > The expression needs to evaluate to "0", "1", "true", or "false", otherwise an error will be thrown.

However, no error is thrown...the block just seems to be skipped.
Documentation also says:
 > The more powerful evaluation capabilities in ROS Jade allow for much more complex expression. Virtually any python expression that evaluates to a Boolean is feasible:

Which is consistent with our observations as in python it's not `true`/`false` but `True`/`False`....but this is consistent with what is said in the first quote.

A little example:
When running `rosrun xacro xacro --inorder` on the following snippet:
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:property name="cond1" value="false"/>
  <xacro:property name="cond2" value="true"/>

  <xacro:if value="${cond1 or cond2}">
    <link name="test"/>
  </xacro:if>

</robot>
```
The resulting URDF looks like this:
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from conditional_block.urdf.xacro   | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
</robot>
```

While running the same command on the following snippet:
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:property name="cond1" value="False"/>
  <xacro:property name="cond2" value="True"/>

  <!-- ros_control plugin -->
  <xacro:if value="${cond1 or cond2}">
    <link name="test"/>
  </xacro:if>

</robot>
```
produces the expected result:
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from conditional_block.urdf.xacro   | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
  <link name="test"/>
</robot>
```

Anybody observed this before?

ipa-mdl, 2017-01-30:
`${cond1 or cond2}` is a python expression, so you have to use `True`/`False` or `=='true'` here.

What happens for `<xacro:if value="${cond1}">`?
If this is working, then everything is fine.

ipa-fxm, 2017-01-30:
The following results in the expected URDF:
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:property name="cond1" value="false"/>
  <xacro:property name="cond2" value="true"/>

  <xacro:if value="${(cond1 == 'true') or (cond2 == 'true')}">
    <link name="test"/>
  </xacro:if>

</robot>
```

Also, 
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:property name="cond1" value="true"/>

  <xacro:if value="${cond1}">
    <link name="test"/>
  </xacro:if>

</robot>
```
returns valid URDFs for all values for `cond1` `true`/`True`/`false`/`False`

Still, it feels a little inconvenient to not be able to intuitively use properties and macro arguments within the python expression of an conditional block - especially when as in most xacro's those properties and macro arguments are set to `true`/`false` as done in the examples of the xacro tutorials...rather than `True`/`False`

rhaschke, 2017-04-24:
The fundamental issue here is the distinction between strings `'true'` / `'false'` and boolean values `True `/ `False`. The property definition 
`<xacro:property name="cond1" value="true"/>`
defines a string property with value `'true'`. Hence, you need to use the more complicated syntax to check for string equality. If you directly want to use Python booleans, you should define the property like this:
`<xacro:property name="cond1" value="${True}"/>`.
However, this currently results in a name-not-found error, which I indeed consider a bug, which I will file elsewhere.

codebot, 2017-04-24:
Fixed in #150, which is now merged. Thanks @ipa-fxm for the detailed report and @rhaschke for the fix!

closed 2017-04-24 by codebot