ros/xacro #148: Running xacro --includes on nested <xacro:include> adds xmlns:xacro tags to <xacro:macro>

mrjogo, 2017-04-14:
Put these files into the same directory (.txt because otherwise GitHub complains):
[main.xacro.txt](https://github.com/ros/xacro/files/921237/main.xacro.txt)
[sublevel1.xacro.txt](https://github.com/ros/xacro/files/921236/sublevel1.xacro.txt)
[sublevel2.xacro.txt](https://github.com/ros/xacro/files/921238/sublevel2.xacro.txt)

If you run `xacro --includes main.xacro.txt > composed.xacro` and then run `xacro --inorder composed.xacro` the second command gives the warning `xacro:macro: unknown attribute(s): xmlns:xacro`.

This is because `xacro --includes` adds the `xmlns:xacro` attribute to any `<xacro:macro>` with a nested `<xacro:include>`.

Ie,
```xml
<xacro:macro name="illustrating_macro">
  <xacro:include filename="sublevel2.xacro.txt" />
...
```
Becomes
```xml
<xacro:macro name="illustrating_macro" xmlns:xacro="http://ros.org/wiki/xacro">
...
```
which apparently does not sit well with `xacro --inorder`

rhaschke, 2017-04-14:
What's the benefit of processing everything twice, first with `--include` option to produce the expanded document and then doing the actual xacro processing?
You are right, I would consider this as a bug too.

mrjogo, 2017-04-14:
I'm combining all the xacro includes into one file, then dynamically combining that with another xacro string before processing it to URDF. It's not a huge deal since it's a warning, but I spent some time tracking it down and wanted to report it to save others the effort.

rhaschke, 2017-11-15:
@mrjogo: Is this issue fixed for you with #151? If so, please close the PR.

PR #151 (2017-08-25): fixes for #148 and #149
Fixes #148 and #149:
- suppress warning about extra attributes that are namespace specifiers
- suppress `--inorder` deprecation warning when `--includes` option is given
commit cd2386ba (2017-08-25): do not complain about extra attributes that are namespace specifiers

fixes #148
commit 795abca2 (2018-03-27): cherry-pick #151

* suppress --inorder deprecation warning when --includes option is given
  fixes #149

* do not complain about extra attributes that are namespace specifiers
  fixes #148
closed 2018-03-27 by rhaschke