ros/xacro #149: xacro --includes gives warning to use --inorder processing, which is not possible

mrjogo, 2017-04-14:
Running the `xacro --includes` command gives the warning

```
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
```

But running `xacro --includes --inorder` gives the error 
```
xacro: error: options --inorder and --includes are mutually exclusive
```

rhaschke, 2017-04-14:
I guess you would like to suppress the deprecation warning when the `--includes` option is used?

mrjogo, 2017-04-14:
If that's the right resolution. I'm unclear if `--includes` somehow breaks the assumptions of `--inorder` processing. Ie, if a file is processed first with `xacro --includes` and then with `xacro --inorder` is it equivalent to just processing it with `xacro --inorder`.

rhaschke, 2017-04-25:
Good point. Usually, `xacro --includes | xacro --inorder` should yield the same result as `xacro --inorder`. 
However, if one uses `${}` expressions for the filename argument of the `<include>` tag, this obviously requires prior (inorder) processing of the document to resolve the variable filename.
But combining both options, `--inorder` and `--include`, to cover this case as well, doesn't make sense: inorder-processing would already modify the document (evaluate expressions + macros), which is not intended by the `--include` option. That's why both options are mutually exclusive.

mrjogo, 2017-04-26:
Ah, I understand now. Thanks!

raequin, 2018-03-23:
Please tell me, if you're able, where I can learn about "traditional processing" vs using the --inorder flag.

rhaschke, 2018-03-26:
Did you have a look into the [wiki](http://wiki.ros.org/xacro#Processing_Order)?

commit 75e8caa5 (2017-08-25): suppress --inorder deprecation warning when --includes option is given

fixes #149
commit 795abca2 (2018-03-27): cherry-pick #151

* suppress --inorder deprecation warning when --includes option is given
  fixes #149

* do not complain about extra attributes that are namespace specifiers
  fixes #148
closed 2018-03-26 by rhaschke