ros/xacro #152: Pass argument and read parameter in xacro file

ipa-mjp, 2017-05-23:
Hello,

- I am trying to call one .xacro file in other xacro file. For example: I call b.xacro in the file of a.xacro.Also I want to pass argument when I call b.xacro file. How can I pass argument? 

- Secondly, b.xacro file I want to read value from parameter. How can I read? In short, I want to do same operation as for calling any function.

Please possible than help me.

Thank you.

guihomework, 2017-05-23:
Hi,
 * calling a macro file in another xacro file => do you mean including ? or calling a macro defined in the included file ?
 * when including I don't think you can pass arguments like a roslaunch include, only the top level a.xacro can use xacro arg to get arguments passed from command line when expanded. 
 * however the xacro properties defined in the a.xacro can be accessed in b.xacro if I remember correctly (take care of the order the properties are created and consumed if using --inorder.
 * default values of macro parameters in b.xacro can use "parent" properties (using '^') see the doc section 6.1

ipa-mjp, 2017-05-24:
Hello,
-Thank you so much for your suggestion.
- Yes, I am talking about calling a macro defined in the included file.

Thank you.

guihomework, 2017-05-24:
Then it is the very standard mechanism of calling the macro. Passing arguments can be done with different means :
* by passing parameters to the macro (section 6 of the doc)
* by using default values for the macro parameters pointing to the parent (section 6.1 of the doc)
* by using properties in a.xacro that are used in b.xacro macro if not defined locally inside the macro in which case namespace and scope must be taken care of. I would not recommend this technique unless you know what you do, as properties might be missing or not defined when the macro is called. With parameters, default values and fallback values you are much safer I would say.

the only difference is that b.xacro contains the macro, and a.xacro includes b.xacro (which defines the macro) and then calls it.

It this solves the issue, please close it. 

You could also use **answers.ros.org** to ask questions, and later create an issue if this is indeed an issue or missing feature of xacro.

rhaschke, 2017-11-15:
@ipa-mjp Is this issue solved for you? If so, please close it.

closed 2017-11-16 by ipa-mjp