ros/xacro #155: xacro problem cannot load the parameter  returned with 2

Avinash439, 2017-08-03:
[image]

guihomework, 2017-08-09:
Seems you miss the package ```fanuc_resources``` which is available here http://wiki.ros.org/fanuc or https://github.com/ros-industrial/fanuc
Adding it should solve your problem (worked for me)
I don't think this is a xacro issue.

rhaschke, 2017-11-15:
@Avinash439 Can this issue be closed?

Avinash439, 2017-11-15:
yes

Best Regards,

Narayana Poorna Naga Avinash
Research Masters Student
Computer Vision and Robotics
Erasmus - VIBOT
University of Burgundy
Le Creusot France
+33 - 07 <+33%207%2051%2001%2017%2051>53834175

closed 2018-03-26 by rhaschke