ros/xacro #168: Adding support for python macros

Zabot, 2017-10-27:
A while back I wrote a python script that generated static xacro macros from a directory of .stls that could be imported into you main .xacro as `hardware.xacro` and significantly simplified the robot description. Unfortunately it was limited in the macros it could generate, because the support for mathematical operations in xacro is limited. We could allow for some more complex xacro macros with some kind of `<xacro:python>` block that functions similarly to a macro block, but replaces the contents with the value returned by a python function, something along these lines:
```python
def mesh_link(name, mesh):
    # Calculate bounding box and inertial tensor from STL
    collision_block = ...
    inertial_block = ...
    return '<link name="{}">{}{}{}</link>'
        .format(name, visual_block, inertial_block, collision_block)
```
```xml
<xacro:python name="mesh_link" name="wheel" mesh="$(find robot_description)/meshes/wheel.stl"/>
```
My mesh property calculation macro is the only concrete use case I've come up with so far, but I could imagine some other more powerful macros that further simplify the process of robot description. What do you all think?

rhaschke, 2017-10-27:
I don't get the full idea yet. Where do you want to place the mesh_link function definition? Inside the .xacro? This is critical, as XML and python are not nicely syntax compatible: You need to escape special chars, care about python indentation, etc.
Also, you mention, that xacro's math capabilities are rather limited. Nowadays you have full python math available. At the time implementing this, we discussed also the security risk of including arbitrary python code snippets, which is raised by your request again.

Zabot, 2017-10-30:
The way I was imagining it the python could either be defined in an external .py file that would be specified in an attribute (either the attribute of the tag itself or perhaps in a previous tag), or it could actually be in the xacro:python block itself, but that might make reusing macros a little weird (And raises all the fun syntax compatibilities you mentioned). Maybe something like this:
```xml
<xacro:python load="($find robot_description)/scripts/macros.py"/>
<xacro:python name="some_macro" some_attr="foo"/>
<xacro:python name="some_other_macro" some_other_attr="bar"/>
```
The parser would import the python modules specified by `load` and then attempt to call functions specified by `name` with the remaining attributes being used as parameters.

I'll have to take another look to see if I can implement things with xacro as it is now. I recall not being able to do sines and cosines to handle rotations, but it's been a while since I looked at it.

In regards to the security issue, given that the python being executed and the xacro calling for its execution are both provided by the user, I don't think there's a significant risk. Unless a user is calling python functions defined in other packages, but then theres always some risk involved in trusting a package creator.

rhaschke, 2018-03-28:
I'm closing this in favour of #170.

closed 2018-03-28 by rhaschke