ros/xacro #169: ModuleNotFoundError: No module named 'fnmatch'

patrickelectric, 2017-11-02:
```
 patrick@dell  ~/ros_ws/lunar  roslaunch urdf_tutorial display.launch model:=urdf/01-myfirst.urdf
... logging to /home/patrick/.ros/log/d6fe2a1a-bfdd-11e7-90c4-f8da0c93ee61/roslaunch-dell-17451.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/mnt/hd/ros_ws/lunar/install_isolated/share/xacro/xacro.py", line 55, in <module>
    import xacro
  File "/mnt/hd/ros_ws/lunar/install_isolated/lib/python3.6/site-packages/xacro/__init__.py", line 35, in <module>
    import glob
  File "/usr/lib/python3.6/glob.py", line 5, in <module>
    import fnmatch
ModuleNotFoundError: No module named 'fnmatch'
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/mnt/hd/ros_ws/lunar/install_isolated/share/xacro/xacro.py urdf/01-myfirst.urdf] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py $(arg model)" name="robot_description"/>
The traceback for the exception was written to the log file
```

rhaschke, 2017-11-16:
Please check your python installation. `fnmatch` is a standard python module.

kartikmohta, 2017-11-16:
@patrickelectric This is a problem with the released version of xacro not being fully compatible with Python 3. This has been fixed in the lunar-devel branch but hasn't been released yet.

patrickelectric, 2017-11-16:
ha ! Ty @kartikmohta, it's working now.

closed 2017-11-16 by patrickelectric