ros/xacro #176: Overrride prorerty from command line

dogoepp, 2018-01-29:
I'm attempting to have two versions of a URDF file to be generated. I went with a a xacro property that would be overridden from the call to xacro, like

```shell
xacro hexaforce.xacro --inorder path_generation:=package > generated.urdf
```

Where hexaforce.xacro is the following file

 ```xml
<?xml version="1.0"?>
<robot name="pexod" xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:property name="path_generation" value="relative" />

  <xacro:if value="${'package' == path_generation}">
    <xacro:property name="path_prefix" value="package://hexapod_description/" />
  </xacro:if>
  <xacro:if value="${'relative' == path_generation}">
    <xacro:property name="path_prefix" value="" />
  </xacro:if>

  <xacro:include filename="$(find hexapod_description)/urdf/modules/hexaforce.xacro" />

</robot>
 ```

However, it seems that the option I give to xacro does not affect the generated file. If the property is not declared in the xacro file, the generation outputs an error.

I can't understand the reason behind this, neither through the documentation nor through tinkering.

guihomework, 2018-01-29:
Hi, 
To pass arguments from the command line  ```xacro:arg``` is what you are looking for. see http://wiki.ros.org/xacro#Rospack_commands

Supposing that you use the global path_prefix in your include, Your file would like this
```
<?xml version="1.0"?>
<robot name="pexod" xmlns:xacro="http://www.ros.org/wiki/xacro">

 <xacro:arg name="path_generation" default="relative" />
 <xacro:property name="path_generation" value="$(arg path_generation)" />

 <xacro:if value="${'package' == path_generation}">
   <xacro:property name="path_prefix" value="package://hexapod_description/" />
 </xacro:if>
 <xacro:if value="${'relative' == path_generation}">
   <xacro:property name="path_prefix" value="" />
 </xacro:if>

<xacro:include filename="$(find hexapod_description)/urdf/modules/hexaforce.xacro" />

</robot>
```

It did not work for me to combine both args and properties in the python expression eval
``` ${'package' == $(arg path_generation)}"```
hence the intermediate properties get the value from the argument.

dogoepp, 2018-01-30:
Hi @guihomework ,
Thanks a lot! It's silly I didn't understand this from the wiki. It works quite well.

closed 2018-01-30 by dogoepp

thebirdgr, 2023-01-10:
@guihomework Thank you! This worked for me!