ros/xacro #181: Cannot install generated urdf files using xacro_install and catkin_make

willcbaker, 2018-03-12:
Using the `xacro_install` macro only works with catkin tools (`catkin build`) and does not work when using `catkin_make install`

Excerpt of CMakeLists.txt
```
# Generate urdf's and put them in the /model/urdf directory
foreach(xacro_file ${model_xacro_files})
  xacro_add_xacro_file(${xacro_file} INORDER)
  list(APPEND xacro_outputs ${XACRO_OUTPUT_FILE})
endforeach()

# Following will install in dev and install folder, does not work with catkin_make
# xacro_install(xacro_target ${xacro_outputs} DESTINATION model/urdf)
# Only installs in install, works with catkin_make
install(DIRECTORY ${URDF_OUTPUT_PATH}
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/model
)
```
When I uncomment my `xacro_install`... line it will only work with catkin tools by calling `catkin build`

rhaschke, 2018-03-16:
Could you, please, provide more details? Running `catkin_make install` on your example, it correctly installs:
```
Install the project...
-- Install configuration: ""
-- Installing: /tmp/ros/install/_setup_util.py
-- Installing: /tmp/ros/install/env.sh
-- Installing: /tmp/ros/install/setup.bash
-- Installing: /tmp/ros/install/setup.sh
-- Installing: /tmp/ros/install/setup.zsh
-- Installing: /tmp/ros/install/.rosinstall
-- Installing: /tmp/ros/install/lib/pkgconfig/yumi_description.pc
-- Installing: /tmp/ros/install/share/yumi_description/cmake/yumi_descriptionConfig.cmake
-- Installing: /tmp/ros/install/share/yumi_description/cmake/yumi_descriptionConfig-version.cmake
-- Installing: /tmp/ros/install/share/yumi_description/package.xml
-- Installing: /tmp/ros/install/share/yumi_description/urdf/yumi.urdf

```

willcbaker, 2018-03-19:
Apparently I was a bit out of date, resolved.

closed 2018-03-19 by willcbaker