ros/xacro #184: deprecating old-style processing, making --inorder the default

rhaschke, 2018-03-28:
Dear @codebot and @wjwwood 

I cleaned up most open issues and PRs. Before going to release, I want to discuss, whether and when we should make the new `inorder` processing the default? For Melodic already? If we decide to switch, I would switch the roles of old and new processing: making old processing available via an option (and urgently asking to adapt the xacro files) and new processing becoming the default (but still being available via `--inorder` as well, which in turn should throw a deprecation warning that the option will be removed in future).

The new processing model is available and deprecated since Jade and we didn't received many complaints on github. However, I think there are still many people out there, still using Indigo or even older ROS releases and thus maybe never learned about the new model.

codebot, 2018-03-28:
Thanks @rhaschke for all of the work cleaning up the issues and merging/porting PRs!

I'm in favor of having `inorder` become the default for Melodic.

wjwwood, 2018-03-28:
Same for me, 👍 for changing the default to `inorder`. Just make sure to note this in the migration guide:

https://wiki.ros.org/melodic/Migration

rhaschke, 2018-03-29:
Thanks for the feedback, @codebot + @wjwwood. Which option name do you suggest for the old behaviour? What about `--legacy`?

codebot, 2018-04-04:
I don't have any great ideas. Maybe something that refers to "order," since that's the legacy behavior that is changing by default? `--legacy-order` maybe? But I don't feel strongly about it. Thanks again for all your hard work on this.

rhaschke, 2018-05-14:
Released 1.13

closed 2018-05-14 by rhaschke