ros/xacro #190: roslaunch Python arg substitution finds wrong package folder path/xacro executable?

skohlbr, 2018-08-17:
I've encountered an issue with using Python arg substitution together with using xacro as described in detail [here on ROS Answers](https://answers.ros.org/question/299232/roslaunch-python-arg-substitution-finds-wrong-package-folder-path/). It leads to only the deprecated `xacro.py` executable being found when using Python arg substitution, while a regular `find` in a launch file finds the `xacro` executable. This probably goes deeper beyond xacro, but @gavanderhoorn suggested posting here first ;)

rhaschke, 2018-08-17:
This is actually intended behaviour for ROS substitution args. I stumbled over this some years ago as well. However I couldn't find the conversation with @dirk-thomas anymore.
The key is, that `$(find pkg)` is not self-contained, but considers the context (stuff after the closing bracket) as well for search. For this reason, `$(find xacro)/xacro` returns a different path than `$(eval find('xacro') + '/xacro')`

closed 2018-08-17 by rhaschke

skohlbr, 2018-08-17:
Thanks for the explanation. So if I understand that right, if I want to use Python substitution args in a launch file, there is no way to get to the non-deprecated `xacro` executable?

rhaschke, 2018-08-20:
>  if I want to use Python substitution args in a launch file, there is no way to get to the non-deprecated xacro executable?

If you insist:
`$(eval find('xacro') + '/../../lib/xacro/xacro --inorder '`