ros/xacro #191: Feature request: '#pragma once' equivalent

JulianAssmann, 2018-10-05:
Often one wants to have one xacro file defining all materials/macros/whatever and include that file in multiple robot component files. Later, you want to combine these components in another file (e. g. 'my_robot.xacro'), that also needs those definitions and therefore includes the components as well as the materials/macros/whatever file. However if you do this, you get errors like

material 'Black' is not unique.

I presume that this is due to indirect multiple inclusions of the materials/macros/whatever files.

It would be nice, if the xacro converter detects these multiple includes and prevents copying the same file content twice, just like the #pragma once statement in C++.

See https://answers.ros.org/question/304797/pragma-once-equivalent-for-urdfxacro-files/ for examples.

rhaschke, 2018-11-01:
@JulianAssmann Can we close this issue?

JulianAssmann, 2018-11-01:
@rhaschke Sure, sorry.

closed 2018-11-01 by JulianAssmann

firesurfer, 2026-02-02:
I think I got that mixed up in my head. I implemented an include guard as described above and it seems to work fine. 
Thanks for your quick response