ros/xacro #192: INORDER by default for xacro_add_files

christian-rauch, 2018-10-13:
The [xacro documentation](http://wiki.ros.org/xacro) states:

> Because --inorder processing is much more powerful, in future versions beyond Jade, the new processing style will become default [...]

However, only `xacro_add_xacro_file` has set INORDER as default:
https://github.com/ros/xacro/blob/0ae956da439bde5dc3fe19d9abc06c618a5ef6e9/cmake/xacro-extras.cmake.em#L37-L39
and there is no parameter that needs to be set:
https://github.com/ros/xacro/blob/0ae956da439bde5dc3fe19d9abc06c618a5ef6e9/cmake/xacro-extras.cmake.em#L14-L15
For `xacro_add_files`, it is still a parameter and needs to be set manually:
https://github.com/ros/xacro/blob/0ae956da439bde5dc3fe19d9abc06c618a5ef6e9/cmake/xacro-extras.cmake.em#L157-L165

For consistency with the documentation, `xacro_add_files` should also use `INORDER` as default.

rhaschke, 2018-10-13:
Thanks for pointing this out. Obviously I missed to update the other cmake function when changing the default.

christian-rauch, 2018-10-14:
The fix solves the issue. Thanks for quickly resolving this.

rhaschke, 2018-10-14:
Resolved by #193

closed 2018-10-14 by rhaschke

130s, 2020-11-25:
Shouldn't this change worth being noted as an API breaker? At minimum in the migration note [wiki.ros.org/noetic/Migration](http://wiki.ros.org/noetic/Migration)?

I run into a build failure when building on Noetic a pkg meant to be built with Melodic
```
CMake Error at /opt/ros/noetic/share/xacro/cmake/xacro-extras.cmake:49 (message):
  no <output> specified for: INORDER
Call Stack (most recent call first):
  /opt/ros/noetic/share/xacro/cmake/xacro-extras.cmake:167 (xacro_add_xacro_file)
  CMakeLists.txt:5 (xacro_add_files)
```

In this [example CMakeLists.txt](https://github.com/osrf/vorc/blob/6f165b0a8fdce2c3281fe59732212d592088ed3d/vorc_gazebo/CMakeLists.txt#L48), simply removing `INORDER` bypassed the build failure.