ros/xacro #195: Override old URDF when calling xacro_add_files

christian-rauch, 2018-10-21:
I am using
```xacro_add_files(${xacro_files} INORDER TARGET media_files INSTALL DESTINATION robots)```
in a `CMakeLists.txt` to convert a bunch of xacro files to urdf when the workspace is built.

The URDFs are place in the same location as the *.urdf.xacro files are, i.e. `path/to/robot.urdf.xacro` -> `path/to/robot.urdf`. But when I change the xacro file and rebuild the workspace, the URDFs are not updated accordingly. I need to remove the previous URDF files before regenerating them when building the workspace.

Is there a way to force the rebuilding of URDFs I.e. let the macro `xacro_add_files` always override the old URDF.

rhaschke, 2018-10-24:
Rebuilding xacros definitely works for me on Melodic. 
Could you provide a minimal workspace exhibiting this issue?

rhaschke, 2018-11-01:
Any news on this? Can we close the issue?

christian-rauch, 2018-11-01:
Oh sorry, I thought I closed the issue already.
I couldn't reproduce the issue on a minimal workspace. It also might have appeared with a different cmake macro. I am going to close this.

closed 2018-11-01 by christian-rauch