ros/xacro #197: ROS Dependencies

fabienrohrer, 2018-10-30:
I'm trying to run Xacro in Travis tests and in a standalone program which aims to run on several OS.
The few dependencies on ROS are very annoying to achieve this goal.

There are very few ROS dependencies, and I'm wondering if fallbacks without ROS could be implemented. Basically these two ones:

1. The substitution mechanism: https://github.com/ros/xacro/blob/melodic-devel/src/xacro/__init__.py#L189

    This is the worst one, because if roslaunch and rospkg are not installed, then xacro fails there.

    Typically arguments for `eval_extension` are `'$(find ur_description)'` and returns a path. Naively, it seems easy to write a working fallback for this case.

2. rosgraph: https://github.com/ros/xacro/blob/0ae956da439bde5dc3fe19d9abc06c618a5ef6e9/src/xacro/cli.py#L96

    Here, it's less problematic, because a non-blocking fallback is already implemented. But I'm wondering about the possible side effects.

What's your point of view on this?

rhaschke, 2018-10-30:
Since e9517623fba0d39b0e3b8a520682ab8815d059bc, xacro is ROS agnostic, i.e. can run without a ROS environment. Of course this implies that you cannot use ROS-specific features in xacro.

> 1. [argument substitution](https://github.com/ros/xacro/blob/melodic-devel/src/xacro/__init__.py#L189)

IMHO argument substitution is inherently coupled to rospkg. If you don't have a ROS environment, you should not use `$(find ...)` and it will work perfectly. 
While you claim, it seems easy to write a fallback, I'm not even sure what this fallback should return semantically. Where do you expect `find` should look for pathes?

> 2. [remapping](https://github.com/ros/xacro/blob/0ae956da439bde5dc3fe19d9abc06c618a5ef6e9/src/xacro/cli.py#L96) 

Again, if you don't have a ROS environment, you shouldn't try to use remappings on the cmdline.

fabienrohrer, 2018-10-30:
Thank you for your quick answer.

That makes sense. The fact is that I do not have necessarily the hand on the xarco file I'm working on, and using `$(find ...)` is a common practice:

https://github.com/ROBOTIS-GIT/ROBOTIS-OP3-Common/blob/master/op3_description/urdf/robotis_op3.urdf.xacro#L4
https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/ur10.urdf.xacro#L9

Maybe a solution for me would be to add a preprocess stage and to search and replace the `$(find )` statements.

rhaschke, 2018-10-30:
> Maybe a solution for me would be to add a preprocess stage and to search and replace the $(find ) statements.

If you know how to replace them, that's a feasible approach. `xacro` cannot know how to replace them.

fabienrohrer, 2018-10-30:
I will give a try. Thank you.

closed 2018-10-30 by fabienrohrer