ros/xacro #209: Unable to use xacro default argument togheter with launch file argument

rickstaa, 2019-05-17:
I am trying to pass a launch file argument to a `xacro` file while also having the ability to set a `xarco:arg` default value when no argument is supplied. I use the solution given in this [ROS answers question](https://answers.ros.org/question/38956/pass-parameters-to-xacro-in-launch-file/) to achieve this. This solution works fine when I supply the launchfile argument to the `xarco` file as follows: 

`    <arg name="myvar" value="0"/>
    <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find franka_description)/robots/panda_arm_hand.urdf.xacro' prefix:=$(arg myvar)"/>
`

Unfortunately, when another third party script does the same call but without supply the `myvar` argument it gets the following error:

`
Undefined substitution argument myvar None                  
`

Which is to be expected since now argument does not exist in the scope of the following `panda_arm_hand.urdf. xacro file`:

```
<?xml version="1.0" encoding="utf-8"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="panda">
  <xacro:arg name="robot_name" default="panda"/>
  <!-- <xacro:arg name="myvar" default="relative" /> -->
  <!-- <xacro:property name="myvar" value="$(arg myvar)" /> -->

  <xacro:include filename="$(find franka_description)/robots/panda_arm.xacro"/>
  <xacro:include filename="$(find franka_description)/robots/hand.xacro"/>
  <xacro:include filename="$(find franka_description)/robots/panda.gazebo.xacro"/>
  <xacro:include filename="$(find franka_description)/robots/panda.transmission.xacro"/>

  <xacro:panda_arm gazebo="$(arg myvar)"/>
  <xacro:hand ns="panda" rpy="0 0 ${-pi/4}" connected_to="panda_link8"/>

  <xacro:panda_gazebo robot_name="$(arg robot_name)" />
  <xacro:panda_transmission robot_name="$(arg robot_name)" />

</robot>
```

I tried to solve this by using the `<xacro:arg name="gazebo" default="relative" /> ` command. Unfortunately, now the value of the myvar variable is set to *relative* instead of *0* when I do supply the `myvar` argument. Is there an easy way to set a xarco default value that is overwritten when an argument with a similar name is supplied in the launch file call?

rhaschke, 2019-05-18:
This works out of the box. Here is a minimal example: 
```xml
<root xmlns:xacro="http://www.ros.org/wiki/xacro">
	<xacro:arg name="prefix" default="default value" />
	$(arg prefix)
</root>
```

`> xacro args.xml:`

```xml
<root xmlns:xacro="http://www.ros.org/wiki/xacro">
  
	default value

</root>
```

`> xacro args.xml prefix:=other`

```xml
<root xmlns:xacro="http://www.ros.org/wiki/xacro">
	other
</root>
```

rhaschke, 2019-05-18:
@rickstaa, your example is inconsistent: From the launch file you pass an argument named `prefix`. However, in your xacro file, the prefix argument is not mentioned. Rather, a `myvar` argument is mention (but commented out).

rickstaa, 2019-05-20:
Dear @rhaschke thanks for your example I got it to work now. You are right when creating the example code I forgot to change `prefix` to `myvar`. I changed it for future reference. My error was because i was using two methods to define arguments at the same time namely:

`<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="panda" parms="myvar:='false'">`

and 

`<xacro:arg name="myvar" default="false" />`

While being inconsistent with the `$(args myvar)` vs `${myvar}` syntax.

closed 2019-05-20 by rickstaa