ros/xacro #216: [ros2] dashing-devel ros-testing debian fails to expand *any* xacro-macros

IanTheEngineer, 2019-10-01:
I transitioned from using the functional https://github.com/ros/xacro/pull/211 to the released Debian ros-testing packages yesterday, and noticed that now while xacro runs, it does not expand **any** macros during its execution:

```
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from /foobar_description/urdf/foobar.urdf.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="foobar">
  <VLP-16 gpu="false" hz="14" name="foobar" parent="robot_base" samples="400" topic="foobar/velodyne_points">
    <origin rpy="2.0 2.0 0.0" xyz="0.0 0.6 0.0"/>
  </VLP-16>
  <sensor_d415 fps="12" parent="base_link" prefix="realsense_a" height="480"  width="720">
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 2.0"/>
  </sensor_d415>
  <sensor_d435 fps="15" parent="base_link" prefix="realsense_b" height="480" width="720">
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 2.0"/>
  </sensor_d435>
...
  <link name="base_link"/>
</robot>
```
The `VLP-16`, `sensor_d415`, and `sensor_d435` lines are all xacro macros and should be fully expanded.

IanTheEngineer, 2019-10-01:
For reference (and my own temporary use), I've forked and pushed up a backup branch of #211 : 
https://github.com/ToyotaResearchInstitute/xacro/tree/ros2-port-find

rhaschke, 2019-10-01:
Unprefixed macro names are deprecated since quite a while and were removed from the ROS2 release completely. Hence, you need to prefix all macros (and all other xacro commands) with `xacro:`. This is also written in the [changelogs](https://github.com/ros/xacro/blob/dashing-devel/CHANGELOG.rst).

rhaschke, 2019-10-03:
Dear @IanTheEngineer, @clalancette, @jacobperron, and other maintainers,

I just noticed that, although deprecated since a while (#79), xacro didn't issue the corresponding warning messages in most cases.
Probably, there are many xacro files out there, which still use unprefixed macro names and xacro tags. Hence, it might be more reasonable to revert this change for now and enable the warning to allow people to adapt their xacro files. What do you think?

jacobperron, 2019-10-03:
> xacro didn't issue the corresponding warning message in most cases.

Out of curiosity, can you elaborate how the warning messages weren't issued?

In any case, the proposal sounds to me. We could even make the warning message explicitly state that support for unprefixed names and tags will be removed in F-turtle.

rhaschke, 2019-10-05:
> Out of curiosity, can you elaborate how the warning messages weren't issued?

I guess the fix 87ca14fb4f29c2d6d21570503ca05228fc1c2565 speaks for itself. Essentially, `deprecated_tag()` wasn't called for macros (but only for xacro keyword tags like `if`, `macro`, `include`) and the static `_issued` flag was always set to `True`.

rhaschke, 2019-10-06:
New release created: https://github.com/ros/rosdistro/pull/22582

closed 2019-10-06 by rhaschke