ros/xacro #224: How to properly handle attribute tag in math expressions?

d-desiatkin, 2019-10-14:
Hello. I have an issue with attribute tag. I want to create a parameter with dynamic name.
Previously I have always created my constants through <xacro:property> tag. Now I wanted to figure out whether I can do it with <xacro:attribute>. I have tried several ideas but not single one worked. Can you clarify the correct usage of xacro attribute?

`<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="porter">

	
	
	<xacro:property name="base_name" value="playing_area" />

	
	<xacro:attribute name="${base_name}_length" value="3.0"/>
	<xacro:attribute name="${base_name}_width" value="3.0"/>
	<xacro:attribute name="${base_name}_thickness" value="3.0"/>
	<xacro:attribute name="${base_name}_mass" value="3.0"/>
	
	
	
	<xacro:property name="ctdp_planks_length" value="1.000" />
	<xacro:property name="ctdp_planks_width" value="0.200" />
	<xacro:property name="ctdp_planks_thickness" value="0.022" />
	<xacro:property name="ctdp_planks_mass" value="3"/>
	
	<xacro:property name="ctdp_top_length" value="0.400" />
	<xacro:property name="ctdp_top_width" value="0.200" />
	<xacro:property name="ctdp_top_thickness" value="0.022" />
	<xacro:property name="ctdp_top_mass" value="3"/>
	

	
</robot>`

`<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="porter">

	
	

	<xacro:macro name="pc_inertia" params="mass length width thickness">
		<inertial>
			<mass value="${mass}" />
			<inertia ixx="${1/12*mass*(width*width+thickness*thickness)}"
				ixy="0.0" ixz="0.0"
				iyy="${1/12*mass*(length*length+thickness*thickness)}" 
				iyz="0.0"
				izz="${1/12*mass*(length*length+width*width)}" />
		</inertial>
	</xacro:macro>
	
	
	
	
	<xacro:macro name="ctdp_plank" params="prefix *position *orientation">
 	
	<link name="${prefix}_ctdp_plank">
		<visual>
			<geometry>
				<box size="${ctdp_planks_length} ${ctdp_planks_width} ${ctdp_planks_thickness}"/>
			</geometry>
			<xacro:insert_block name="orientation" />
			<material name="white"/>
		</visual>
		<collision>
			<geometry>
				<box size="${ctdp_planks_length} ${ctdp_planks_width} ${ctdp_planks_thickness}"/>
			</geometry>
			<xacro:insert_block name="orientation" />
		</collision>
		<xacro:pc_inertia mass="${ctdp_planks_mass}" length="${ctdp_planks_length}" width="${ctdp_planks_width}" thickness="${ctdp_planks_thickness}"/>
	</link>
 
	<joint name="${base_name}_to_${prefix}_ctdp_plank" type="fixed">
		<parent link="${base_name}"/>
		<child link="${prefix}_ctdp_plank"/>
		<xacro:insert_block name="position" />
	</joint>
	
	</xacro:macro>
	
</robot>`

`<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="field">
	<xacro:include filename="$(find field_model)/xacro/parameters.xacro"/>
	<xacro:include filename="$(find field_model)/xacro/macroses.xacro"/>

	
	<link name="${base_name}">
		<visual>
			<geometry>
				<box size="${base_name}_length ${base_name}_width ${base_name}_thickness"/>
			</geometry>
			<origin rpy="0 0 0" xyz="0 0 0"/>
			<material name="blue"/>
		</visual>
		<collision>
			<geometry>
				<box size="${base_name}_length ${base_name}_width ${base_name}_thickness"/>
			</geometry>
		</collision>
		<xacro:pc_inertia mass="${base_name+'_mass'}" length="${base_name+'_length'}" width="${base_name+'_width'}" thickness="${base_name+'_thickness'}"/>
	</link>
	

 	
	
</robot>`

rhaschke, 2019-10-14:
`xacro:attributes` are intended for setting dynamically named XML attributes.
```xml
<a xmlns:xacro="http://www.ros.org/wiki/xacro">
	<xacro:property name="base_name" value="base" />
	<tag><xacro:attribute name="${base_name}_pi" value="3.14"/></tag>
</a>
```
gives:
```xml
<a>
	<tag base_pi="3.14"/>
</a>
```
I hope this clarifies the usage. If not, please provide a _minimal_ example and pose a concrete question.

d-desiatkin, 2019-10-14:
Thank you for the quick response. 
Now I get the idea. It would be great to add this example to http://wiki.ros.org/xacro#Elements_and_Attributes

closed 2019-10-14 by d-desiatkin