ros/xacro #225: Substitution arg for $(find pkg)

Levaru, 2019-10-17:
Hi, I'm currently at a roadblock with this issue. Is there a way to dynamically find directory paths for packages according to xacro args that i pass into the file via roslaunch? Maybe something like the eval command from roslaunch?

This works in a launch file:

```
<arg name="robot_name" default="kr60" />
<arg name="robot_path" default="$(eval find( arg('robot_name')+'_description'))" />

<!-- Load urdf description -->
<param name="robot_description" command="$(find xacro)/xacro '$(arg robot_path)/urdf/$(arg robot_name).urdf.xacro'"/>
```

But the same thing doesn't work in a xacro file:
```
<xacro:arg name="robot_name" default="kr60" />
<xacro:arg name="robot_path" default="$(eval find( arg('robot_name')+'_description'))" />

<!-- Load description modules -->
<xacro:include filename="$(arg robot_path)/urdf/$(arg robot_name).xacro" />
```

Is eval only available for roslaunch? Would it be possible to also include it in xacro?

rhaschke, 2019-10-17:
No, we don't have `$(eval ...)` in `xacro`. But you can use a property within `$(find)`:
```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/xacro" name="robot" xacro:targetNamespace="http://www.ros.org">
	<xacro:property name="robot_name" value="panda" />
	$(find ${robot_name}_moveit_config)
</robot>
```

closed 2019-10-17 by rhaschke

martinandrovich, 2021-10-16:
@rhaschke Is it possible to nest `$(find)` and `$(arg)` without the definition of a `<xacro:property />`? Something like:
```
...value="$(find $arg('robot_name'))" ...
```