ros/xacro #230: Python 3 status?

mikepurvis, 2019-12-06:
Looks like a `noetic-devel` branch exists, but it's not the default, no release has been cut, and a handful of commits have gone into `melodic-devel` that haven't been ported forward:

https://github.com/ros/xacro/compare/noetic-devel...melodic-devel

What's the state of affairs here?

rhaschke, 2019-12-08:
I merged melodic-devel changes into noetic-devel and tried to prepare a noetic release. But rosdep complains about several not-yet-released packages:
```
==> git-bloom-generate -y rosdebian --prefix release/noetic noetic -i 1 --os-name ubuntu
Generating source debs for the packages: ['xacro']
Debian Incremental Version: 1
Debian Distributions: ['bionic']
Releasing for rosdistro: noetic

Pre-verifying Debian dependency keys...
Running 'rosdep update'...
Could not resolve rosdep key 'catkin'
Failed to resolve catkin on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'catkin', aborting.
catkin is depended on by these packages: ['xacro']
<== Failed
Could not resolve rosdep key 'roslaunch'
Failed to resolve roslaunch on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'roslaunch', aborting.
roslaunch is depended on by these packages: ['xacro']
<== Failed
Could not resolve rosdep key 'roslint'
Failed to resolve roslint on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'roslint', aborting.
roslint is depended on by these packages: ['xacro']
<== Failed
Could not resolve rosdep key 'rostest'
Failed to resolve rostest on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'rostest', aborting.
rostest is depended on by these packages: ['xacro']
<== Failed
```
@sloretz, from https://github.com/ros/rosdistro/issues/21513 I guess you are going to become the noetic release maintainer.
Can you shortly summarize the state? Do you have a build farm setup already?

sloretz, 2019-12-09:
> Can you shortly summarize the state? Do you have a build farm setup already?

The Noetic buildfarm is not yet set up. The current practice is to just add a source entry. That allows others to use `rosinstall_generator` to create a Noetic workspace and build from source.

rhaschke, 2019-12-09:
What about the missing dependencies (catkin, roslaunch, roslint, rostest). Is there a reason that they are not yet released/available in the distro?

sloretz, 2019-12-09:
[catkin](https://github.com/ros/rosdistro/blob/master/noetic/distribution.yaml#L33), [roslaunch, and rostest](https://github.com/ros/rosdistro/blob/3a03796331ea6feadb5796069e6d8463f373ef38/noetic/distribution.yaml#L273) have source entries. `roslint` does not: https://github.com/ros/roslint/issues/70.

Bloom releases for Noetic don't work yet; source entries have to be made manually. IIUC, bloom releases are blocked until the bootstrap Ubuntu-F apt repo contains things like `catking_pkg`, `rospkg`, etc.

mikepurvis, 2019-12-09:
Oh ha, `roslint` is on me— I'll take care of it momentarily.

mikepurvis, 2019-12-18:
@rhaschke Please also change the default branch.

rhaschke, 2019-12-18:
I will not change the default branch, before Noetic isn't be released.
Most changes probably apply to the melodic branch as well and I prefer to merge them into the noetic branch from there.

closed 2019-12-18 by rhaschke

mikepurvis, 2019-12-18:
Okay, cool— as long as there's a plan.

With `ros_comm` we've tended to do a final batch of merges to the present devel branch *before* forking off the new devel branch, and then targeting everything to that with cherry-picks in only one direction. But the volume of changes here is a lot lower, so picking in multiple directions is perhaps less likely to result in a chaotic mess. :)