ros/xacro #236: xacro give a  ` unknown attribute(s)` warning while include a xacro file with arguments

lianghongzhuo, 2020-04-24:
xacro give a  ` unknown attribute(s)` warning while include a xacro file with arguments
as discussed in https://github.com/TAMS-Group/tams_ur5/issues/9
we find that in [this line](https://github.com/TAMS-Group/tams_ur5/blob/master/tams_ur5_description/urdf/arm.urdf.xacro#L11) if we include a xacro file with arguments the xacro will give a warning `xacro:include: unknown attribute(s): prefix`.

v4hn, 2020-04-24:
The unknown attribute is clearly our own bug. The actual error to be reported here, however, lies in the error output of xacro:

```
when processing file: *src/tams/tams_ur5/tams_ur5_description/urdf/ur5.urdf.xacro
included from: */tams/tams_ur5/tams_ur5_description/urdf/arm.urdf.xacro
included from: */src/tams/tams_ur5_setup/tams_ur5_setup_description/urdf/tams_ur5_setup.urdf.xacro
xacro:include: unknown attribute(s): prefix
```

The first line is plain wrong.

The `xacro:include` that triggers the warning is in `arm.urdf.xacro` and the included file is `robotiq-3f-gripper_articulated_macro.xacro`.

`ur5.urdf.xacro` was the last included file in `arm.urdf.xacro` right before the problematic include.
So I guess the processing variables referring to the current file is not updated correctly?

rhaschke, 2020-04-24:
Maybe, handling errors in xacro:include (which is a corner case) is not yet handled correctly.

rhaschke, 2020-04-24:
I cannot reproduce the error message. For me, it doesn't show any file context.
So, please provide more information: 

1. Which xacro version you are using?
2. What are your cmdline arguments?

lianghongzhuo, 2020-04-26:
1 I use the master branch
2. Run `roslaunch tams_u5_bringup tams_ur5.launch` can reproduce the error message.

lianghongzhuo, 2020-04-27:
Sorry, this is my fault. The error message is for a previous warning for other things, not relevant to the `prefix` bug we introduce.
So I will close this issues.
But maybe we can add some more information when giving `unknown attribute(s):` warning, to tell the user which xacro file has the wrong use of xacro.

rhaschke, 2020-04-27:
Reopening as a reminder to provide more information for (all) warnings.

closed 2020-07-05 by rhaschke