ros/xacro #240: Adding "dotted dict" access to yaml loaded dictionaries?

gavanderhoorn, 2020-05-29:
The yaml support in xacro has been great, it allows for some really flexible `.xacro`s, but one thing always bothered me: with deeply nested dicts, readability really suffers.

Example:

```xml
<limit effort="${settings['my_robot']['my_joint']['limits']['effort']}"
  lower="${settings['my_robot']['my_joint']['limits']['pos_low']}"
  upper="${settings['my_robot']['my_joint']['limits']['pos_up']}"
  velocity="${settings['my_robot']['my_joint']['limits']['vel']}" />
```

A contrived example perhaps, but close to reality.

A work-around has been to extract values and assign them to `xacro:property`s, but that is really more of a work-around and essentially trade readability for having a lot of `xacro:property` lines (with long names).

It would be nice perhaps to access yaml loaded dicts using dotted notation, which would get us:

```xml
<limit effort="${settings.my_robot.my_joint.limits.effort}"
  lower="${settings.my_robot.my_joint.limits.pos_low}"
  upper="${settings.my_robot.my_joint.limits.pos_up}"
  velocity="${settings.my_robot.my_joint.limits.vel}" />
```

still long, but definitely more readable.

Would a PR adding this be accepted into `xacro`?

This doesn't have to be implemented by overriding `__getattr__(..)` on the actual `dict`, but could also be added as an extension of the parser which would split the variable identifier at the `.` and then use the resulting tokens to index into the yaml-loaded dict.

rhaschke, 2020-05-30:
In such cases as you described, we usually define a local (property) variable:
```xml
<xacro:property name="limits" value="${settings['my_robot']['my_joint']['limits']}" />
```
and then reuse it like this:

```xml
<limit effort="${limits['effort']}" lower="${limits['pos_low']}" upper="limits['pos_up']}" velocity="${limits['vel']}" />
```

But I agree that your suggested notation is more elegant. 
However, the code in `${}` is interpreted directly by python:
https://github.com/ros/xacro/blob/1e3a5b29cb2d97377604cb0a2cc77d08185a8bd4/src/xacro/__init__.py#L525-L527

So I would not mess up further with the (very simple) parser, but indeed overriding `__getattr__()` of the dict returned by `load_yaml()`. In principle, this needs to be done recursively. However, we might want to do it lazily, i.e. only when really accessing a particular sub dict.

rhaschke, 2020-05-30:
I missed to explicitly answer your most important question: Yes, I would merge this feature :wink:

gavanderhoorn, 2020-06-30:
As always, this has been pre-empted by other work :(

I'll try to get to it at a later time though.

@rhaschke: I noticed that namespace resolution already used "dotted notation".

Would that be something to look at, or is that handled by the parser -- which you mentioned should not be extended any further?

rhaschke, 2020-06-30:
I just implemented a first draft of this feature in #240. Please have a look if that already suits your needs.
If so, may I ask you to augment a unit test function for it as well? Somewhere close to:
https://github.com/ros/xacro/blob/1e3a5b29cb2d97377604cb0a2cc77d08185a8bd4/test/test_xacro.py#L1126-L1138

Thanks.

gavanderhoorn, 2020-07-01:
> I just implemented a first draft of this feature in #240. Please have a look if that already suits your needs.

That's really nice of you @rhaschke. I did not write https://github.com/ros/xacro/issues/240#issuecomment-651768731 with the intention of getting anyone else to do this.

I'll try to take a look at the tests.

rhaschke, 2020-07-05:
Implemented in #245.

closed 2020-07-05 by rhaschke