ros/xacro #248: YamlDictWrapper breaks dict functionality

dustingooding, 2020-08-08:
This commit: https://github.com/ros/xacro/commit/d1171d06d0ebef15eac246ca1b6ef24c3314c9c1

The new `YamlDictWrapper` has broken normal Python `dict` usage.  The ability to see if a key exists in a dictionary object is now impossible.  Given this dict value:

```
d = {foo: 'bar'}
```

The following xacro snippets throw exceptions and prevent execution:

```
<xacro:if value="${baz in d}">
```
```
<xacro:property name="abc" value="d['foo'] if 'foo' in d else 'xyz'"/>
```

I don't feel that dot accessors are worth destroying the concept of a dictionary and all the features they have.  I feel `YamlDictWrapper` should be reverted.  If that's not acceptable, perhaps make `YamlDictWrapper` "isA" dict instead of "hasA" dict?

rhaschke, 2020-08-08:
Thanks for pointing this out. I would have preferred to inherit `YamlDictWrapper` from `dict` in first place. However, I couldn't figure out how to turn an existing `dict` (return from yaml parsing) into a `YamlDictWrapper` inherited from dict - without copying all the data. That's why we have chosen the `hasA` approach. Any ideas to implement the `isA` approach? I'm welcoming a PR.

dustingooding, 2020-08-08:
Yeah, I'm working on a PR that removes `YamlDictWrapper`.  dot accessors aren't required.

dustingooding, 2020-08-08:
https://github.com/ros/xacro/pull/249

rhaschke, 2020-08-09:
Sorry, but removing the wrapper functionality isn't a valid solution to the problem. I was asking for alternative solutions.

dustingooding, 2020-08-09:
It's perfectly valid solution.  It restores functionality that was broken by an unnecessary feature.  I don't have time to fix a half-baked "improvement".  I'm just going to roll-back and compile from source.

rhaschke, 2020-08-09:
Fixed via #250.

PR #250 (2020-08-09): Inherit YamlDictWrapper from dict
Fixes #248.
closed 2020-08-09 by rhaschke