ros/xacro #254: YamlDictWrapper screws up yaml lists

rhaschke, 2020-09-09:
Hi @gavanderhoorn,
we came across another regression introduced by YamlDictWrapper (#245): A top-level list returned by `load_yaml()` is passed to the dict constructor, which of course fails. Consider this simple xacro example:
```xml
<a xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:property name="lists" value="${load_yaml('list_of_lists.yaml')}"/>
  ${lists[0]} 
</a>
```
```yaml
- A
- B
```
which fails with: 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: dictionary update sequence element #0 has length 1; 2 is required
```
A simple fix (#253) just ignores non-dicts for wrapping at top-level as well. However, this raises another question: 
What about dicts that are part of lists? Do we want to provide dotted access for those as well? IMO, this would require much more effort, because we would need to provide a `YamlListWrapper` as well to propagate the wrapper down to any element that might get accessed. What do you think? Is that worth the effort?

gavanderhoorn, 2020-09-10:
I haven't looked into this issue in sufficient detail to think about a work-around/fix, but for now I'd mitigate it by removing the dict wrapper.

rhaschke, 2020-10-13:
Addressed via #258.

PR #258 (2020-10-13): yaml loading: Recursively wrap lists and dicts for dotted dict access
Fixes #254. Rebase of #253.
closed 2020-10-13 by rhaschke