ros/xacro #256: Caching property evaluations or not?

rhaschke, 2020-09-29:
Consider the following xacro snippet, that defines the two properties `a` and `x`. Due to historical reasons (handling non-`inorder` evaluation), we decided to cache the values of properties once they got evaluated.
```xml
<root version="1.0" xmlns:xacro="http://www.ros.org/wiki/xacro">
	<xacro:property name="a" value="2"/>
	<xacro:property name="x" value="${2*a}"/>
	<A x="${x}"/>

	<!-- redefining a (but not x) and re-evaluating x yields the old, cached value of x -->
	<xacro:property name="a" value="4"/>
	<B x="${x}"/>

	<!-- redefining x as well, resets the cache and triggers new evaluation -->
	<xacro:property name="x" value="${2*a}"/>
	<C x="${x}"/>
</root>
```
Thus, currently the above yields:
```xml
<root version="1.0">
  <A x="4"/>
  <B x="4"/>
  <C x="8"/>
</root>
```
To handle #255, I need to rework property evaluation for Noetic/python3 anyway. For this reason, I want to get some feedback on what result people expect from the above xacro snippet. Particularly, @codebot, @wjwwood, @gavanderhoorn: What do you think?

rhaschke, 2020-09-29:
A colleague of mine argued that the "caching" behaviour is what one would expect from a programming language:
```python
a = 2
x = 2*a
a = 1
x should be still 4
```
I think, what I actually stumbled over was that we have both lazy evaluation and caching:
```xml
<root version="1.0" xmlns:xacro="http://www.ros.org/wiki/xacro">
	<!-- evaluation of x is postponed if not yet possible, storing the expression instead of the value -->
	<xacro:property name="x" value="${2*a}"/> <!-- a is not yet defined -->
	<xacro:property name="a" value="2"/>
	<A x="${x}"/> <!-- only here, x is finally evaluated, storing the value for future use -->

	<!-- redefining a (but not x) and re-evaluating x yields the old, cached value of x -->
	<xacro:property name="a" value="4"/>
	<B x="${x}"/>

	<!-- redefining x as well, resets the cache and triggers new evaluation -->
	<xacro:property name="x" value="${2*a}"/>
	<C x="${x}"/>
</root>
```

wjwwood, 2020-09-29:
I agree that the output is expected. I think the value of `x` is determined when declared, i.e. the statement `${2*a}` is evaluated immediately. If you want it to be evaluated lazily that would be more like a function, where `${x}` is calling the function "x" to have it evaluated at that point.

I agree that mixing the "caching" and optional lazy evaluation is really weird. But I can see how the "caching" is least surprising while the optional lazy evaluation might be convenient for how some xacro files are written.

Personally, I'd vote for your second example raising an exception when `<xacro:property name="x" value="${2*a}"/> ` occurs, but I can't imagine what issue that might cause downstream.

gavanderhoorn, 2020-09-30:
I believe I would agree with @wjwwood: lazy evaluation can be a nice feature to have, but it should not be done without the user being aware of it, for risk of running into situations which you already describe @rhaschke.

Personally I've always treated `xacro` as just-another-programming-language. Macros just happen to 'return' snippets of urdf or xacro. I believe the `--inorder` evaluation makes the most sense.

But that's probably because I"m used to programming languages, their structure and behaviour.

I'd not be happy with changes which would alter the behaviour of `xacro` and move it away from that model, but I'm of course only one voice here.

rhaschke, 2021-02-15:
Thanks for the confirmation of current behavior.

closed 2021-02-15 by rhaschke