ros/xacro #263: Issue with include evaluation of filename if namespace is used

Doomerdinger, 2021-02-10:
**example.xacro**
```xacro
<?xml version="1.0" ?>
<robot name="example" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:property name="include_file" value="$(find example_package)/child.xacro"/>

  <xacro:include filename="${include_file}" ns="child_ns"/>
</robot>
```
The above xacro file will fail, complaining that `include_file` is not defined. This error does not present itself if the `ns` attribute is not used.

**example2.xacro**
```xacro
<?xml version="1.0" ?>
<robot name="example" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:arg name="include_file" default="$(find example_package)/child.xacro"/>

  <xacro:include filename="$(arg include_file)" ns="child_ns"/>
</robot>
```
The above file appears to work just fine.

rhaschke, 2021-02-11:
Thanks for reporting this issue. I filed a fix in #264.

rhaschke, 2021-02-11:
Closed via #264.

PR #264 (2021-02-11): Use outer-scope symbols to resolve include filename in xacro:include
Fixes #263.
closed 2021-02-11 by rhaschke