ros/xacro #265: Cannot construct dicts with numerical leading names

Doomerdinger, 2021-02-11:
I expect to be able to construct a property dict that can (at least on a basic level) mimic a yaml file.

```xacro
<xacro:property name="offset" value="${dict(1x=0)}"/>
```
results in the following error:
```
invalid syntax (<string>, line 1) 
when evaluating expression 'dict(1x=0)' 
when evaluating expression 'offset'
```

If you try to mark the name as a string like so
```xacro
<xacro:property name="offset" value="${dict('1x'=0)}"/>
```
you get the following error
```
keyword can't be an expression (<string>, line 1) 
when evaluating expression 'dict('1x'=0)' 
when evaluating expression 'offset'
```
Also applies if ` is used.

This makes it impossible to replicate a yaml file using dicts if it has keys that start with numbers, like so
```yaml
---
1x: 0
```

Leading numerical values in dict keys is perfectly valid in Python and in yaml, I expect this not working in xacro dict is a bug and not an intentional decision.

rhaschke, 2021-02-11:
Also in Python, you cannot create such a dict using the constructor syntax:
```python
dict('1x'=0)
SyntaxError: expression cannot contain assignment, perhaps you meant "=="?
```
or
```python
dict(1x=0)
SyntaxError: invalid syntax
```
It only works via curly braces:
```python
{'1x': 0}
```
However, unfortunately, this interferes with the simple parser of xacro, which cannot accept other curly braces in a `${...}` expression. I'm afraid there is no solution to your problem.

rhaschke, 2021-02-11:
Actually, there is another dict constructor using lists, which works: 
```python
dict([('1x',0)])
```

Doomerdinger, 2021-02-11:
You're certain it is not possible? I don't quite see how the dict can be limited as such, but loading a yaml file will work just fine. 

I'd be happy to look into it a bit myself though I am not currently familiar with the codebase. I won't bother if you are certain nothing can be done.

Doomerdinger, 2021-02-11:
> Actually, there is another dict constructor using lists, which works:
> 
> ```python
> dict([('1x',0)])
> ```

Huh! Noted. I was not aware of that format, I'll go ahead and use that for now.
Not sure this bug is worth tackling (if possible) if there exists an alternative, considering this is probably a niche issue.

May be good to make note of these forms on the wiki?

rhaschke, 2021-02-11:
You are welcome to augment the wiki.

Doomerdinger, 2021-02-12:
I'll put it on my agenda.

closed 2021-02-12 by Doomerdinger