ros/xacro #277: How to assign xacro arguments to xacro macro paramters ?

adbidwai, 2021-07-17:
I am initialising a macro in my xacro file as
 `<xacro:macro name="sensor_d435" params="name:=realsense_d435_camera topics_ns:=realsense_d435 parent *origin">`. 
 
I have a xacro argument with the name `robot_namespace` in my launch file. I want to assign it to the `topics_ns` while initializing my macro, something like: `topics_ns:=$(arg robot_namespace)`. When I do this and launch my original launch file, I get an error saying `invalid expression: $(arg`

Probably, it is seeing `$(arg` and `robot_namespace)` as 2 different and separate parameters. 
Any way how I can achieve what I want?

TIA

guihomework, 2021-07-18:
Hi, 

We do exactly what you suggest, we passe launch file arguments to the instantiation of the xacro file see here
https://github.com/ubi-agni/agni_robots/blob/melodic-devel/agni_description/launch/arm_upload.launch

if it did not help, you should provide a small example to reproduce the error.

adbidwai, 2021-07-19:
Hi @guihomework Thanks a lot for your answer 
But in your xacro, you are passing your launch file _argument_ to your xacro _argument_. I want to pass my launch file _argument_ to my xacro _**parameter**_. 

`<xacro:macro name="sensor_d435" params="name:=realsense_d435_camera topics_ns:=realsense_d435 parent *origin">`
So in the above case, I have an argument called `robot_namespace` initialized in my xacro file and want to assign it to one of the macro parameters. 
So I wish to do something like this (but that's not working :( )
`<xacro:macro name="sensor_d435" params="name:=realsense_d435_camera topics_ns:=$(arg robot_namespace)/realsense_d435 parent *origin">`

guihomework, 2021-07-19:
Hi, 

the doc states :
Xacro currently supports all the rospack commands that roslaunch supports using substitution args. Arguments **need to be specified on the command line** using the myvar:=true syntax. 

so in order for $(arg) to be used in your xacro file, the _argument_ must be **declared** in your xacro file, AND you need to **pass** it on the command line _argument_ of the xacro command.
In your xacro file those arguments are then extracted using the $(arg) syntax.  With this _argument_ passed to xacro you should be able to do what you want with it, and use it as a _parameter_ of your initialization of your macro.

your xacro instantiation will happen in a lauch file isn't it ? at instantiation, I think you need to pass _arguments_ to the xacro command, and only these _argument_ can reach the xacro file.

it is like chaining two command line commands: 
launch command line args, extracted by launch syntax, and passed to xacro command line args, extracted by xacro syntax.
otherwise how can a xacro file guess command line arguments if they are not passed at the command line as argument ?

if your problem is a feature request, I strongly suggest to give a full example launch file and xacro file so that the developer can look it up.

rhaschke, 2021-07-19:
I can confirm that the following xacro file fails with the error `invalid expression: $(arg`
```xml
<robot name="example" xmlns:xacro="http://www.ros.org/wiki/xacro">
	<xacro:arg name="foo" default="foo" />
	<xacro:macro name="test" params="arg:=$(arg foo)">
		${arg}
	</xacro:macro>

	<xacro:test/>
</robot>
```
Looks like argument evaluation doesn't work within macros, which is definitely a bug. I will look into this asap.

rhaschke, 2021-07-19:
You ran into a limitation of the macro argument parsing. Essentially the naive parser splits argument items at spaces. Thus, there are two arguments extracted:
- `arg` := `$(arg`
- `foo)`

Currently, you need to work around that limitation by defining a helper property:
```xml
<robot name="example" xmlns:xacro="http://www.ros.org/wiki/xacro">
	<xacro:arg name="foo" default="foo" />
	<xacro:property name="foo" value="$(arg foo)" />
	<xacro:macro name="test" params="arg:=${foo}">
		${arg}
	</xacro:macro>

	<xacro:test/>
</robot>
```

You are welcome to improve the regex expression here to handle arg and property expressions as well, i.e. `$( )` and `${ }`:
https://github.com/ros/xacro/blob/e07841aa5ed4e41e113bfa30bceeac1013c923a9/src/xacro/__init__.py#L491-L492

guihomework, 2021-07-19:
@adbidwai sorry for not having understood your exact problem. I initially wanted to save @rhaschke some time but I failed.

adbidwai, 2021-07-19:
@guihomework No problem Sir, it definitely provided some insights
@rhaschke @guihomework Thank you for reviewing this issue and thanks @rhaschke for creating a PR for this feature!

rhaschke, 2021-07-19:
Fixed via #278.

PR #278 (2021-07-19): Improve macro arg parsing
... supporting:
- `$(substitution args)`
- `${python expressions}`
- `'single quoting'` or `"double quoting"` of spaces

Fixes #277. @adbidwai, please review.
closed 2021-07-19 by rhaschke