ros/xacro #280: [melodic-only] name 'sorted' is not defined in macro argument.

v4hn, 2021-09-03:
I ran into this trying to run https://github.com/captain-yoshi/scene_parser (as dependency of https://github.com/captain-yoshi/moveit_benchmark_suite) on melodic.

xacro 1.14.8 (noetic) supports this piece of code:
```
<?xml version="1.0" ?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="test">
  <xacro:macro name="foo" params="sorted_keys:=^">
    <xacro:property name="xyz" value="${sorted_keys}"/>
  </xacro:macro>

  <xacro:foo sorted_keys="${sorted([2,1,0])}"/>
</robot>
```
1.13.12 (melodic) on the other hand fails with

    name 'sorted' is not defined
    when evaluating expression 'sorted([2,1,0])'

At first glance it looks like https://github.com/ros/xacro/pull/278 should have fixed this and was released in 1.13.*12*.
Is the failure in melodic intended or is this an unresolved bug in the branch?

rhaschke, 2021-09-03:
Actually, it is a bug in Noetic, that `sorted()` works! I'm looking into the issue.

captain-yoshi, 2021-09-03:
My use case for `sorted` is for looping a map which you can only loop as a list in Xacro (I think). I could change my map to a list but then I loose the ability to get the data by name. Maybe there are other ways to do it without using sorted.

```yaml
apcshelf:
  subframes:
    binA:
      xyz: [0.10479, -0.291029, 1.52393]
    binB:
      xyz: [0.10479, 0, 1.52393]
```

rhaschke, 2021-09-03:
Sorry, my statement was misleading: I plan to keep `sorted` both in Melodic and Noetic. However, the (current) availability of `sorted` in Noetic pointed me to another issue.
@captain-yoshi: Could elaborate more about your use? What do you mean by "looping a map"? I'm curious how you "unfold" the loop. Can you provide the xacro expression you are using?

v4hn, 2021-09-03:
> Can you provide the xacro expression you are using?

the full macro I based the minimal example  on can be found [here](https://github.com/captain-yoshi/scene_parser/blob/f5d8801fffc9bebaddae3429a22e92a238b215fc/config/utils.xacro#L25).

rhaschke, 2021-09-03:
Fixed via 2a130a93adcd593e53d6906958fcf5087ac5d39c.
Releases triggered into [Melodic](https://github.com/ros/rosdistro/pull/30692), [Noetic](https://github.com/ros/rosdistro/pull/30693), [Foxy](https://github.com/ros/rosdistro/pull/30695), [Galactic](https://github.com/ros/rosdistro/pull/30696), [Rolling](https://github.com/ros/rosdistro/pull/30697)

closed 2021-09-03 by rhaschke

rhaschke, 2021-09-03:
> The full macro I based the minimal example on can be found [here](https://github.com/captain-yoshi/scene_parser/blob/f5d8801fffc9bebaddae3429a22e92a238b215fc/config/utils.xacro#L25).

That's so cool. I never thought about calling the macro recursively to unroll a loop!

@captain-yoshi, you can even simplify a little more. You are right, you can only loop over a list in this fashion. But you could easily pass the list (of map values) as `items="${list(dict(a=1, b=2, c=3, d=4, e=5).values())}"`:
```xml
<robot name="loop" xmlns:xacro="http://www.ros.org/wiki/xacro">
	<xacro:macro name="loop" params="items:=^">
		<xacro:if value="${items}">
			<!-- pop first item from list -->
			<xacro:property name="item" value="${items.pop(0)}"/>

			<item>${item}</item>

			<!-- recursively call myself -->
			<xacro:loop/>
		</xacro:if>
	</xacro:macro>
	<xacro:loop items="${[1,2,3,4,5]}"/>
</robot>
```

v4hn, 2021-09-04:
Thanks for the impressively fast debugging and sound bugfix+issue fix! :1st_place_medal:

captain-yoshi, 2021-09-06:
Thank you very much Robert !!