ros/xacro #281: ros2 branch is broken

brennand, 2021-09-13:
Hello, 

The current head of ros2 is broken, 2.0.7.  When I try and use the UR_description that has been running for the last couple of  months it will no longer convert these degrees to anything and I get an error

```
joint_limits:
  shoulder_pan:
    # acceleration limits are not publicly available
    has_acceleration_limits: false
    has_effort_limits: true
    has_position_limits: true
    has_velocity_limits: true
    max_effort: 150.0
    max_position: !degrees  180.0
    max_velocity: !degrees  180.0
    min_position: !degrees -180.0
```

The error:

`Use of invalid name(s):  __limits 
when evaluating expression '__limits['shoulder_pan']['min_position']'
`

The urdf:

`    <xacro:property name="__joint_limit_parameters" value="${load_yaml(joint_limits_parameters_file)}"/>
    <xacro:property name="__kinematics_parameters" value="${load_yaml(kinematics_parameters_file)}"/>
    <xacro:property name="__physical_parameters" value="${load_yaml(physical_parameters_file)}"/>
    <xacro:property name="__visual_parameters" value="${load_yaml(visual_parameters_file)}"/>

    
    <xacro:property name="__limits" value="${__joint_limit_parameters['joint_limits']}"/>
    <xacro:property name="__dh_parameters" value="${__physical_parameters['dh_parameters']}"/>
    <xacro:property name="__offsets" value="${__physical_parameters['offsets']}"/>
    <xacro:property name="__inertia_parameters" value="${__physical_parameters['inertia_parameters']}" />
    <xacro:property name="__mesh_files" value="${__visual_parameters['mesh_files']}" />
    <xacro:property name="__kinematics" value="${__kinematics_parameters['kinematics']}" />

    
    <xacro:property name="shoulder_pan_lower_limit" value="${__limits['shoulder_pan']['min_position']}" scope="parent"/>`

When I switch the branch to this: 5671b7d (2.0.7) it all works fine.

rhaschke, 2021-09-14:
Since 2.0.7 xacro forbids the use of double-underscore names for security reasons.
See https://github.com/ros-industrial/universal_robot/pull/579 for a fix of your issue.

closed 2021-09-14 by rhaschke

gavanderhoorn, 2021-09-28:
Just to clarify: the ROS 2 version of `ur_description` is somewhat decoupled from `ros-industrial/universal_robot`.

The issue was fixed there in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/166.